ros2/ros2_documentation #3898: [Iron] Does the tutorial of "Creating and using plugins (C++)" only support Linux?

songyuc, 2023-09-02:
Hi, guys!

I am a learner of Iron, and I notice that in the ROS Iron tutorial of "Creating and using plugins (C++)", the first command is for Linux other than Win10.
So, I am wondering whether this tutorial is applicable to the Windows 10 system?

Your answer and guide will be appreciated!

clalancette, 2023-09-05:
No, it works on all of our platforms.  See #3905 which corrects this.

closed 2023-09-05 by clalancette

songyuc, 2023-09-06:
Hi, @clalancette, I checked the [ROS Iron tutorial of “Creating and using plugins (C++)”] page, and it still only provides the [[sudo apt-get]](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Pluginlib.html#prerequisites) command for installing pluginlib. How can I install this pluginlib on Win10?

clalancette, 2023-09-06:
> Hi, @clalancette, I checked the [ROS Iron tutorial of “Creating and using plugins (C++)”] page, and it still only provides the [[sudo [...]

We only rebuild the documentation overnight.  It should now be updated.  In short, if you have successfully installed ROS 2, you don't need to do anything further; pluginlib is part of the core.

songyuc, 2023-09-06:
@clalancette , thanks very much for your explanation.

PR #3905 (2023-09-05): Fix up the Pluginlib tutorial to not be Linux only.
The Prerequisites made it seem like this was a Linux only tutorial, when in reality it works on all of our platforms. Change this, and also do some light editing while we are in here.

Fixes #3898