ros2/ros2_documentation #3910: [ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'turtle_tf2_py' not found, searching: ['D:\\\\launch\\\\launch_ws\\\\install', 'C:\\\\dev\\\\ros2_iron']"

songyuc, 2023-09-11:
Description:
I'm following the [Managing large projects](https://docs.ros.org/en/iron/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html), and trying to run the launch file `launch_turtlesim_launch.py` in my ROS 2 environment, but I'm getting an error that says "package 'turtle_tf2_py' not found". I want to try installing the [turtle_tf2_py](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html) package. And I want to know whether turtle_tf2_py support Windows?

Steps to reproduce:
1. Navigate to the directory containing `launch_turtlesim_launch.py`.
2. Run the command `ros2 launch launch_tutorial launch_turtlesim_launch.py`.

Expected behavior:
The launch file should run successfully and start the turtlesim simulation.

Actual behavior:
The launch file fails with the error message "package 'turtle_tf2_py' not found".

Environment:
- ROS 2 version: Iron

clalancette, 2023-09-11:
Did you follow the instructions in http://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html to install it from source?

songyuc, 2023-09-15:
I have installed `turtle_tf2_py`. And I rerun the command `ros2 launch launch_tutorial launch_turtlesim_launch.py`. But the turtles in the two turtlesim simulations did not move as shown in the tutorial. What should I do?

clalancette, 2023-09-15:
> I have installed `turtle_tf2_py` from source. And I rerun the command `ros2 launch launch_tutorial launch_turtlesim_launch.py`. But the turtles in the two turtlesim simulations did not move as shown in the tutorial. What should I do?

Please open up a question on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

With that, I'm going to close this out as I don't think there is anything to change in the documentation for this.  If you find out that we need a documentation change, please feel free to reopen.

closed 2023-09-15 by clalancette

songyuc, 2023-09-15:
Hello [@clalancette](https://github.com/clalancette) and ROS 2 Documentation Team,

Firstly, thank you for your previous response. I appreciate the suggestion to post on Stack Exchange, but I believe that providing the complete project code for the ["Managing large projects"](https://docs.ros.org/en/iron/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html) tutorial could be beneficial for everyone.

I've been trying to follow the tutorial step-by-step, but I've encountered issues that I suspect might be related to the tutorial itself or to running it on a Windows 10 system. Having access to the complete project code would allow me to test it directly and determine if the issues are inherent to the code or specific to my setup.

Would it be possible for you to share the complete project code for this tutorial? This would not only help me but also others who might be facing similar issues.

Given the ongoing challenges and the potential benefit of resolving them for a wider audience, I kindly request that you consider reopening this issue for further discussion.

Thank you for your time and consideration.

clalancette, 2023-09-15:
> Would it be possible for you to share the complete project code for this tutorial? This would not only help me but also others who might be facing similar issues.

What is in the tutorial is the complete project code.  There is nothing additional, as far as I am aware.  If there are improvements you'd like to make specifically for Windows, then please feel free to open a pull request with those updates.  We are happy to review them.

songyuc, 2023-09-15:
Dear [@clalancette](https://github.com/clalancette),

I've been going through the ROS 2 documentation, specifically the tutorial on [Using ROS 2 Launch For Large Projects](https://docs.ros.org/en/iron/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html). While the tutorial is informative, I find some parts of the code examples to be a bit unclear.

For instance, the tutorial mentions, "Now let’s create a new `turtlesim_world_3_launch.py` file similar to `turtlesim_world_2_launch.py` to include one more `turtlesim_node` node." As a beginner, I'm not entirely sure what "similar" entails in this context. I feel that the `turtlesim_world_3_launch.py` file could have a lot of code that could be shown to make things clearer.

I believe that providing a more detailed example for `turtlesim_world_3_launch.py` would greatly benefit beginners like myself. Could you please consider adding more explicit code examples to clarify this part of the tutorial?

Thank you for your time and consideration. I look forward to your response.

songyuc, 2023-09-21:
Dear [@clalancette](https://github.com/clalancette),

I hope this message finds you well. I wanted to reach out to you regarding a problem I am facing regarding this tutorial. I have posted a question on the Robotics Stack Exchange platform, specifically related to running the "turtle_tf2_py" package on Windows. The question can be found at this link: [https://robotics.stackexchange.com/questions/104187/windows-issues-running-turtle-tf2-py-package-in-ros-2-environment](https://robotics.stackexchange.com/questions/104187/windows-issues-running-turtle-tf2-py-package-in-ros-2-environment).

I am seeking your assistance in resolving this issue. Your expertise and knowledge in ROS would be greatly appreciated in helping me find a solution.

Thank you in advance for your time and support.

Best regards.

rencmbr, 2024-02-27:
Hello,

I am also following the Intermediate tutorials step by step. The "Managing Large Projects" tutorial is in the "Launch" group of tutorials. These tutorials come BEFORE the tf2 group of tutorials but already use the turtle_tf2 package. Even if the "Managing Large Projects" includes in its Prerequisites section the information that it depends on turtle_tf2_py, I think that it should include  a reminder to install turtle_tf2 in section "4. Reusing Nodes" with a link to the install instructions. This would avoid the errors that @songyuc reported and I also had when doing the tutorials in sequence.

With my best regards,
Renato.