ros2/ros2_documentation #3922: Humble - Compotition Demo Launch link points to the wrong file name

edgarcamilocamacho, 2023-09-16:
Here:

https://github.com/ros2/ros2_documentation/blob/humble/source/Tutorials/Intermediate/Composition.rst?plain=1#L197

It points to a file that doesn't exist:

https://github.com/ros2/demos/blob/humble/composition/launch/composition_demo_launch.py

The correct one is:

https://github.com/ros2/demos/blob/humble/composition/launch/composition_demo.launch.py

I see that in `Iron` and `Rolling` the file name was changed instead of the documentation, but I understand the standard is `.launch.py` instead of `_launch.py`, is it not?

clalancette, 2023-09-17:
> I see that in `Iron` and `Rolling` the file name was changed instead of the documentation, but I understand the standard is `.launch.py` instead of `_launch.py`, is it not?

Well, we kind of moved away from the `.launch.py` standard, and now prefer the `_launch.py` standard, which is why Iron and Rolling have that.  But you are right that it should be `.launch.py` in Humble.  See #3923

edgarcamilocamacho, 2023-09-17:
@clalancette thanks for the information. Not sure if this is the correct place to discuss that, but, where can I find the reason why `_launch.py` is preferred over `.launch.py`?

clalancette, 2023-09-18:
> @clalancette thanks for the information. Not sure if this is the correct place to discuss that, but, where can I find the reason why `_launch.py` is preferred over `.launch.py`?

To be perfectly frank, I don't remember why we decided that way.  But I do know that we had some one way and some the other way, so we decided to make everything consistent and we happened to choose `_launch.py`.

Closing since we merged #3923, but feel free to keep commenting.

PR #3923 (2023-09-18): [humble] Change link to composition_demo.launch.py
Use the older name, since that is what is shipped in Humble.

This will fix #3922
closed 2023-09-18 by clalancette