ros2/ros2_documentation #3948: ]Attribution in contributed documentation pages

mhidalgo-bdai, 2023-10-02:
This is more of a question than an issue (

mhidalgo-bdai, 2023-10-02:
FYI @jbarry-bdai

clalancette, 2023-10-02:
> This is more of a question than an issue. We are looking forward to contribute changes we have made internally, extending existing documentation pages. Attributing those changes to our organization would simplify the bureaucracy on our side, but we do not know if that is even a thing. So we ask you, @clalancette @audrow, how do you handle this, if at all?

We typically only do attribution based on the Signed-off-by line in the commit that adds it.  You are also free to add some additional metadata to those commits, but I don't think that we should start attributing the documentation pages themselves to one organization or another; they are meant to be a shared (and shared responsibility) resource.

mhidalgo-bdai, 2023-10-03:
Got it. Question answered. Thanks @clalancette !