ros2/ros2_documentation #3960: Issue: Undefined 'now' variable in 'turtle_tf2_listener.py' tutorial

songyuc, 2023-10-09:
I am attempting to follow along with the [Using Time (Python) — ROS 2 Documentation tutorial](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html). In the "1. Update the listener node" section, the tutorial instructs to edit `turtle_tf2_listener.py` and remove the `timeout=Duration(seconds=1.0)` parameter from the `lookup_transform()` call, and gives the following code:
```python
trans = self._tf_buffer.lookup_transform(
to_frame_rel,
from_frame_rel,
now)
```
However, upon inspecting the `turtle_tf2_listener.py` source file, I noticed that the `now` variable is not defined anywhere in the file. Because of this, I am unable to successfully follow along with the tutorial.

I wanted to ask if this is an error in the tutorial, or if the source file has been updated and the tutorial has not? How should I proceed in order to successfully complete the tutorial?

I am using ROS 2 Iron.

Thank you very much for your assistance.

jkaniuka, 2024-02-17:
Hi @songyuc, your suspicions were correct :wink: - source file was updated, but tutorial has not. You can simply replace `now` with `rclpy.time.Time()`. See details [here](https://github.com/ros/geometry_tutorials/commit/2479342bd52979b9ec4faa4969d9428291c89189#diff-c3299e618a727c9b027e7d2b754fc135b5121cc6af788f07848880346a7d957dL67-R69) :robot:

clalancette, 2024-05-07:
We "fixed" this by removing the tutorial in #4384, so closing.

closed 2024-05-07 by clalancette