ros2/ros2_documentation #3986: Reading from a bag file (C++) : small incoherence with basic tutorial ROS2 bag

julienthevenoz, 2023-10-25:
Hey people,
I'm new to ROS2 and using all the great tutorials to learn. I just wanted to point out that in the advanced tutorial "Reading from a bag file (C++)" there is a slight incoherence with the basic tutorial it builds upon.
Indeed in the prerequisites section it says "This tutorial discusses using ROS 2 bags. You should have already completed the [basic ROS 2 bag tutorial], and we will be using the **subset** bag you created there." 
The incoherence is that the C++ Reader tries to read messages from the topic /turtle1/pose, but in the basic tutorial ROS2 bag tutorial where we create the **subset** bag, we record the /turtle1/cmd_vel topic. So if you do the two tutorials in a row and try to run, the C++ Reader won't show anything in the console (which confused me for a few minutes).
The simple fix is of course to replace "/turtle1/pose" with "/turtle1/cmd_vel" in the C++ Reader and everything works.
Also the final line before the summary should then say "You should see the (x, y) velocity commands of the turtle printed to the console." instead of "the (x,y) coordinates".

clalancette, 2023-10-26:
> The incoherence is that the C++ Reader tries to read messages from the topic /turtle1/pose, but in the basic tutorial ROS2 bag tutorial where we create the **subset** bag, we record the /turtle1/cmd_vel topic.

As far as I can tell, we record both the `/turtle/cmd_vel` topic as well as the `/turtle1/pose` topic; see http://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html#record-multiple-topics .

Did you have something else in mind?

clalancette, 2024-10-11:
Closing with no response in a year.

closed 2024-10-11 by clalancette