ros2/ros2_documentation #3993: [Windows] Clarification on Supported Systems for URDF Tutorial

songyuc, 2023-10-27:
Hello,

I am currently going through the [Using URDF with Robot State Publisher](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html) tutorial. In one of the steps, the command `mkdir -p ~/second_ros2_ws/src` is given. However, this command seems to be tailored for Unix-like systems and does not work on Windows.

I am using ROS 2 Iron on a Windows machine and I am wondering if this tutorial is suitable for Windows systems. If it is, could you please provide the equivalent commands for Windows or mention the supported systems at the beginning of the tutorial?

I believe clarification on the supported systems or alternative commands for Windows would be beneficial for other users as well.

Thank you!

clalancette, 2023-11-08:
> I am using ROS 2 Iron on a Windows machine and I am wondering if this tutorial is suitable for Windows systems. If it is, could you please provide the equivalent commands for Windows or mention the supported systems at the beginning of the tutorial?

Yes, it should work for Windows as well.  The commands as given should work fine for Windows.  Can you give an example of what is not working for you?

songyuc, 2023-11-09:
Hi, @clalancette 
I don't think it a severe problem, as it would need deleting the "install" folder before running it.

By the way, there is a ["source install/setup.bash"](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#:~:text=source%20install/setup.bash) command, which seems not to work on Windows.

clalancette, 2023-11-09:
> By the way, there is a ["source install/setup.bash"](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#:~:text=source%20install/setup.bash) command, which seems not to work on Windows.

Yeah, that's fair.  I also notice there are a couple of `mkdir` in there, which should be `md` on Windows.

If you'd like to open a pull request that adds in tabs for Windows support (using `md` and `call setup.bat`), I'd be happy to review it.

songyuc, 2023-11-09:
OK. I might firstly try to reimplement this tutorial and then see what I can do.

songyuc, 2023-11-20:
Hi, @clalancette

I have created a PR that aims to address the Windows compatibility issues discussed in this issue.

The PR https://github.com/ros2/ros2_documentation/pull/4024 adds Windows command modifications and supporting documentation to the URDF tutorial.

Changes include:
- Replacing `mkdir` with `md`
- Calling `setup.bat` instead of `source setup.bash`
- Updated documentation on supported platforms

Could you please take a look and provide feedback on the proposed changes? Your review would help improve the PR.

Let me know if any part of the implementation or documentation needs further clarification. I appreciate you taking the time to review this. Please feel free to suggest any other improvements as well.

Thank you in advance for your assistance in getting this contribution merged!

closed 2023-11-23 by songyuc