ros2/ros2_documentation #3996: [Windows] Issues with Running URDF Tutorial of  the Building a Visual Robot Model with URDF from Scratch tutorial on Windows

songyuc, 2023-10-30:
I have been following the [Building a Visual Robot Model with URDF from Scratch](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html) tutorial using ROS 2 on Windows. I encountered an issue when executing the command to launch the `display.launch.py` file in the [One Shape](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html#one-shape) section:
```bash
ros2 launch urdf_tutorial display.launch.py model:=urdf/01-myfirst.urdf
```

The terminal returned a file not found error: `[WinError 2] The system cannot find the file specified.` even though the `01-myfirst.urdf` file exists in the `urdf` directory within the `urdf_tutorial` package. I attempted the command modification suggested in the tutorial to work regardless of the current working directory, but it returned an unrecognized argument error.
```bash
ros2: error: unrecognized arguments: --share urdf_tutorial`/urdf/01-myfirst.urdf
```

I would like to know if this tutorial is supported on Windows and if there are additional steps required to run it successfully on this platform. Any help or guidance would be greatly appreciated.

Thank you!

mjcarroll, 2023-11-09:
It may be that this tutorial needs to be updated to take advantage of file-content substitution, which was added late in the iron release cycle: https://github.com/ros2/launch/pull/708

songyuc, 2023-11-13:
Hi, @mjcarroll,

If possible, I would greatly appreciate your assistance in updating the tutorial to address these Windows-specific challenges. Additionally, if there are any suggestions or temporary workarounds I could try in the meantime, I would be grateful for your guidance.

Thank you for your time and dedication to maintaining the ROS documentation. Your efforts greatly aid learners like myself in navigating and utilizing ROS more effectively.

Looking forward to your response.

Best regards,
Yucheng Song

sloretz, 2023-11-16:
@clalancette is it possible ros2/launch#708 needs a backport? 🧇

clalancette, 2023-11-16:
> @clalancette is it possible [ros2/launch#708](https://github.com/ros2/launch/pull/708) needs a backport? 🧇

Well, we already backported it to iron a while ago.  We never did backport it to humble, though I don't think that is what this issue is asking for (since it was filed against iron).

That said, in the meantime we actually have removed the problematic portion of this documentation.  In particular, `urdf_tutorial` now depends on `urdf_launch`, which has some of this functionality built-in.  So it is possible that nothing more needs to be done here, and that can be used directly.

@songyuc Can you give this another try on Windows?

songyuc, 2023-11-18:
Please give me more time to conduct testing.