ros2/ros2_documentation #4022: IDEs page for ROS2

RobertoRoos, 2023-11-17:
The old wiki has a list of tips for using different IDEs: http://wiki.ros.org/IDEs

Maybe it would be could to migrate this to the ROS2 documentation?

I just went through something of a struggle to get PyCharm to work (https://robotics.stackexchange.com/questions/105462) and I would like to add this to a general list.

clalancette, 2023-11-17:
We have one (old) page for Eclipse at http://docs.ros.org/en/rolling/Tutorials/Miscellaneous/Building-ROS2-Package-with-eclipse-2021-06.html .

But I agree that it would be nice to have a list of IDEs, probably as a "How-to Guide".  I don't have any immediate plans to work on this, but if you'd like to contribute a new page in "source/How-To-Guides/IDEs.rst" with your PyCharm experience, I'd be happy to review it.

songyuc, 2023-11-23:
I look forward to your PR as a PyCharm user.

PR #4023 (2023-11-27): Added new IDEs page and started filling it in for PyCharm
To fix #4022 , as a migration of http://wiki.ros.org/IDEs.

I'm not sure yet if we should move the current content of the VSCode/Docker How-To into this page, or we keep just a link to it.
closed 2023-11-27 by clalancette