ros2/ros2_documentation #4037: [Windows & Iron] Clarification on Windows Support for Fast DDS Discovery Server Tutorial

songyuc, 2023-11-28:
Hi, guys,

I am currently working through the ["Using Fast DDS Discovery Server as discovery protocol"](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#launch-listener-node) tutorial in the ROS 2 Iron documentation. I encountered a step where it's required to set an environment variable using the `export` command (specifically `export ROS_DISCOVERY_SERVER=127.0.0.1:11811`), which is not supported in the Windows CMD or PowerShell environment.

Could you please clarify if this tutorial is intended to be compatible with Windows systems? If so, could you provide guidance or alternate instructions for Windows users for these specific steps?

I believe that including Windows-specific instructions could be highly beneficial for a broader audience trying to follow this tutorial.

Thank you for your attention to this matter.

Best regards,
Yucheng Song

sloretz, 2023-12-08:
@MiguelCompany or @EduPonz do either of you know if the Fast DDS discovery server is usable on windows? That is, [should we fix the tutorial](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#launch-listener-node), or mark it as linux only?

MiguelCompany, 2023-12-11:
@sloretz Fast DDS discovery server is usable on windows. Using `set ROS_DISCOVERY_SERVER=...` instead of `export ROS_DISCOVERY_SERVER=...` would be enough for the tutorial to work on that platform

songyuc, 2023-12-19:
By the way, I found the network traffic generation script provided in the section "Compare Fast DDS Discovery Server with Simple Discovery Protocol" to be very useful for understanding the performance differences between the two discovery protocols.

However, I noticed that the script is a `.bash` script, which is not directly compatible with the Windows PowerShell environment that I am working in. I would like to request a version of the script (or an additional script) that can be run in Windows PowerShell. This would be greatly helpful for Windows users who are following the ROS 2 documentation and want to perform similar network traffic analysis.

Thank you for considering this request. I am looking forward to your response.

Best regards,
Yucheng Song

closed 2023-12-26 by clalancette