ros2/ros2_documentation #4057: Query: Applicability of Gazebo Simulation Setup Tutorial on Windows

songyuc, 2023-12-09:
Hello, guys,

I was going through the [Setting up a robot simulation (Gazebo)](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html) tutorial in the ROS 2 Iron documentation and I am interested in setting up a robot simulation environment on a Windows system.

I noticed that the tutorial does not explicitly mention support for Windows. Could you please confirm if this tutorial is applicable to Windows, or if there are any additional steps or modifications required to follow it on a Windows platform?

Thank you for your assistance.

Best regards,
Yucheng Song

clalancette, 2023-12-11:
I believe that the older version of Gazebo referenced in that tutorial doesn't support Windows.  But maybe @azeey can chime in here and say whether it would be supported?

azeey, 2023-12-11:
> I believe that the older version of Gazebo referenced in that tutorial doesn't support Windows. But maybe @azeey can chime in here and say whether it would be supported?

Yes, I believe that's correct. However, starting in Gazebo Garden, you should be able to use WSL in Windows (see https://github.com/gazebosim/gz-sim/issues/920). If you need to install Gazebo Garden or Harmonic with ROS 2, please make sure to follow https://gazebosim.org/docs/harmonic/ros_installation.

Running Gazebo on Windows is not fully supported yet (see https://github.com/gazebosim/gz-sim/issues/2089)

songyuc, 2023-12-12:
Hello, @azeey,

I'm interested to know if there is a roadmap, plan, or timeline for making Gazebo Simulator compatible with Windows systems instead of WSL in the future. As a Windows user, I'm keen to explore Gazebo's capabilities on this platform and would appreciate any information regarding the development efforts in this regard.

Thank you!

azeey, 2023-12-12:
Windows support is currently community driven, i.e, we rely on PRs submitted by users other than the core simulation team that works on Gazebo. Therefore, there is no solid timeline for fully supporting Gazebo on Windows. That being said, we are very receptive to any contributions that moves this forward.

closed 2023-12-19 by songyuc