ros2/ros2_documentation #4069: [Windows & Iron] AttributeError in ros2 security create_keystore Command

songyuc, 2023-12-19:
Hello, guys,

I encountered an error while following the ROS 2 Security tutorial, specifically when executing the `ros2 security create_keystore` command. The error details are as follows:

**Environment:**

-   ROS 2 Version: Iron
-   Operating System: Win10
-   Python Version: 3.8.3

**Command Executed:**
```
ros2 security create_keystore demo_keystore
```

**Error Message:**
```
D:\dev\ros\sros2_demo>ros2 security create_keystore demo_keystore
Traceback (most recent call last):
  File "C:\dev\ros2_iron\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.25.2', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_iron\Lib\site-packages\ros2cli\cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\command\security.py", line 36, in main
    return extension.main(args=args)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\verb\create_keystore.py", line 38, in main
    sros2.keystore.create_keystore(args.ROOT)
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\keystore\_keystore.py", line 82, in create_keystore
    _utilities.create_smime_signed_file(
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\_utilities.py", line 67, in create_smime_signed_file
    f.write(_sign_bytes(cert, private_key, content))
  File "C:\dev\ros2_iron\Lib\site-packages\sros2\_utilities.py", line 140, in _sign_bytes
    flags = SSLBinding.lib.PKCS7_DETACHED
AttributeError: module 'lib' has no attribute 'PKCS7_DETACHED'
```

I followed the steps outlined in the [ROS 2 Security tutorial](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Introducing-ros2-security.html). This error seems to suggest an issue with the SSL library in the `sros2` package.

Could you please advise on how to resolve this issue or provide any known workarounds?

Thank you!

songyuc, 2023-12-19:
This issue has been mentioned on [[ros2/sros2]](https://github.com/ros2/sros2) as https://github.com/ros2/sros2/issues/285.

clalancette, 2023-12-20:
I'm going to close this out as a duplicate of https://github.com/ros2/sros2/issues/285

closed 2023-12-20 by clalancette