ros2/ros2_documentation #4079: [Windows & ros2-humble-20231122-windows-release-amd64]

samdarwisch, 2023-12-23:
Hallo,

I am trying to install ROS 2 on windows 11 . I went through the setup guide without any issues 
i recive this Message

"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually

i have installed all the nessecary packages but when i come to the step :
Start a command shell and source the ROS 2 setup file to set up the workspace:

call C:\dev\ros2_humble\local_setup.bat"

I don't get what is missing. Do I need to execute "call" in a specific folder or do I lack a package? I am quite blind in this area and this is a rookie question. Thus, if I may, please be detailed in your answer ;)

PLEASE Help, it is Project for the Universita and i will be preciate if someone contact me on DIscord [aramia1614]

littlemiddlebiddle, 2023-12-26:
If you go to:

C:\dev\ros2_humble\

(in my case it was C:\dev\ros2_iron\)

and run:

`pip install packaging
`
then run the example:

`ros2 run demo_nodes_cpp talker`  

does that fix it for you?  I was getting an error before I installed packaging and then, after I did that, it runs for me (I didn't change anything else).  I have a suspicion that they actually *are* referring to that error: 

""[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually"

when they say you should see this error once: 

“The system cannot find the path specified.” 

I could be wrong, but that is my suspicion.  Let me know if you had success with that (sorry, no discord).

clalancette, 2023-12-26:
> "[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually

In short, you don't need to worry about this warning.  It means that you don't have Connext installed, but that is not a requirement.  Just go ahead with the rest of the tutorials.

closed 2024-01-04 by clalancette