ros2/ros2_documentation #4095: Inquiry about Windows support for examining network traffic tutorial

songyuc, 2024-01-09:
Hello ROS 2 Documentation maintainers,

I was going through the [Examining network traffic tutorial](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Examine-Traffic.html) and I was wondering if this tutorial is supported on Windows. The tutorial mentions the use of `tcpdump`, which to my knowledge is not available on Windows.

Is there an alternative method or tool recommended for Windows users to follow this tutorial? Any guidance or additional documentation on this would be highly appreciated.

Thank you for maintaining the documentation and for your support!

Best regards,
Yucheng Song

clalancette, 2024-01-19:
Yeah, this one is very Linux-specific.  We should probably put a note at the top saying that.

songyuc, 2024-01-22:
Hi, @clalancette,
Could you please help put this note?

I appreciate your help very much in advance!

PR #4128 (2024-01-24): Mark the "Examine-traffic" tutorial as Linux specific.
While we are in here, also notice that we use the word "guide" in ROS2-Tracing-Trace-and-Analyze, and switch that to "tutorial".

Fixes #4095
closed 2024-01-24 by clalancette