ros2/ros2_documentation #4103: Tutorial mentions license argument twice in call to `ros2 pkg create`

ivanperez-keera, 2024-01-15:
The following line in the documentation instructs users to pass the license argument to `ros2` twice:

https://github.com/ros2/ros2_documentation/blob/913f2b6cfa9f9dd1ae22fc49764b18316204ebe3/source/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.rst?plain=1#L183

The same is true for the python variant a little further below:

https://github.com/ros2/ros2_documentation/blob/913f2b6cfa9f9dd1ae22fc49764b18316204ebe3/source/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.rst?plain=1#L189

When `--license` is passed twice, the second one takes precedence over the first.  I have checked this by running the command on my computer with difference licenses and checking that the first one was ignored:
```
$ ros2 pkg create --build-type ament_cmake --license Apache-2.0  --node-name my_node test_echo --license BSD-3-Clause
going to create a new package
package name: test_echo
destination directory: /home/myuser/ros_tests/src
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['me <my_email@my_domain.com>']
licenses: ['BSD-3-Clause']
build type: ament_cmake
dependencies: []
node_name: my_node
creating folder ./test_echo
creating ./test_echo/package.xml
creating source and include folder
creating folder ./test_echo/src
creating folder ./test_echo/include/test_echo
creating ./test_echo/CMakeLists.txt
creating ./test_echo/src/my_node.cpp
$ grep -nHre 'Apache' 
$ grep -nHre 'BSD' 
test_echo/package.xml:8:  <license>BSD-3-Clause</license>
```

ivanperez-keera, 2024-01-16:
@clalancette Was this closed by my PR? It appears that was merged but the merge commit didn't close this.

clalancette, 2024-01-16:
Closing now.

closed 2024-01-16 by clalancette