ros2/ros2_documentation #4168: Add support for Algolia in search

AbhijithGanesh, 2024-02-14:
# Algolia search

The current full-text search supported by Sphinx really misses out on edge cases, I have brought this up on the discourse thread: https://discourse.ros.org/t/growing-issue-with-ros-documentation/36075

I feel we could use the Algolia search to improvize the same

kscottz, 2024-02-15:
If you are going to request a feature you need to at least point to the tool you propose we use.  My cursory Google search indicates that [Algolia is a proprietary AI search tool that costs approximately $0.40 per 1000 searches](https://www.algolia.com/pricing/). We have around 100,000 visitors a month. How do you propose we pay for this?

AbhijithGanesh, 2024-02-15:
I have renamed the issue to "better search" so that we could navigate through all open-source solutions that fit in our issue

AbhijithGanesh, 2024-02-20:
@kscottz we need to decide how to move with this! I've marked my solutions in the thread, all of them will need considerable moving of the docs to a new framework