ros2/ros2_documentation #4178: Port GDB Guide from NAV2 to ros2_documenation

Ryanf55, 2024-02-20:
# Purpose

NAV2 has an excellent guide on using GDB with ROS 2:

https://navigation.ros.org/tutorials/docs/get_backtrace.html

The guide has very little to do with NAV2, and is helpful for anyone using ROS. 

This ticket is to keep track of porting the information over. 

# Improvements

1. Use the `--mixin debug` when compiling rather than modifying the CMakeLists to add debug symbols.
2.  Link in some GDB training like this one - using GDB is more than getting a backtrace:  https://www.youtube.com/watch?v=PorfLSr3DDI
3. Perhaps link in using VSCode instructions - the statement `Many IDEs will have some kind of debugger or profiler built in, but with ROS, there are few IDEs to choose.` is not as true with open source editors such as VSCode and most debugging is done on dev machines anyways

fujitatomoya, 2024-02-20:
@Ryanf55 i think this is useful for developers, IMO probably we can move the ROS 2 generic part from Nav2 to ROS 2 doc? So that Nav2 just has a link to that? (Nav2 specific guide can stay in Nav2 repo.)

Ryanf55, 2024-02-20:
Yes, that is what I had in mind.

Ryanf55, 2024-02-20:
Can we tag this as " good first issue" and hopefully someone new gets inspired to do the work?

GauravKumar9920, 2024-02-23:
hii, can I work on this issue ?

fujitatomoya, 2024-02-24:
@GauravKumar9920 thank you very much, i will be happy to review.

GauravKumar9920, 2024-02-25:
Shall, I commit the Tutorial first, followed by the improvements in the future commits.
Also any recommendations on which ROS package example must I use to elaborate on this point.

> 4. Provide an example repository with a segfault that users can try GDB out on, and perhaps integrate the backward-cpp interface.

GauravKumar9920, 2024-02-25:
Further, under what hierarchy of the ROS2 documentation shall I add this section in ?

Ryanf55, 2024-02-25:
> Shall, I commit the Tutorial first, followed by the improvements in the future commits. Also any recommendations on which ROS package example must I use to elaborate on this point.
> 
> > 4. Provide an example repository with a segfault that users can try GDB out on, and perhaps integrate the backward-cpp interface.
> 
> p.s - some general examples have already been included in the documentation.

Yep, that sounds like a good plan. 

For which package, it does not matter. It might actually be a good idea to create a ROS package that intentionally has a segfault, so others can reproduce it.

As far as where, I might put it under intermediate:
https://docs.ros.org/en/humble/Tutorials/Intermediate.html

Ryanf55, 2024-03-06:
Looks like there's other resources from Autoware explaining similar things.
https://autowarefoundation.github.io/autoware-documentation/main/support/troubleshooting/#died-process-issues

clalancette, 2024-04-21:
Closed by #4190

closed 2024-04-21 by clalancette