ros2/ros2_documentation #4184: Codespace doesn't let me `make html`

audrow, 2024-02-22:
When I do `make html`, I get a locale error. I'm not sure what the fix is, especially because I can't seem to use `sudo`, perhaps because of the Docker setup.

```
$ make html
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/rosindex/.local/lib/python3.10/site-packages/sphinx/__main__.py", line 7, in <module>
    raise SystemExit(main(sys.argv[1:]))
  File "/home/rosindex/.local/lib/python3.10/site-packages/sphinx/cmd/build.py", line 326, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python3.10/locale.py", line 620, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
make: *** [Makefile:23: html] Error 1
```

Any thoughts, @cychitivav?

It also may be good to add some instructions saying how to use the codespace in the repo's readme.

fujitatomoya, 2024-03-02:
as @clalancette pointed out, we need to add some procedure during devcontainer instantiation. i will come up with fix 🔨

clalancette, 2024-10-11:
I think this was fixed by @fujitatomoya 's PRs, so I'm going to close it out.  Feel free to reopen if this is still an issue.

closed 2024-10-11 by clalancette