ros2/ros2_documentation #4188: launch file undocumented "args" option?

stewpend0us, 2024-02-26:
in my yaml I can do:
```
...
- node:
    ...
    args: "--ros-args --log-level debug"
```

but I haven't been able to find this documented anywhere? Is this the best way/only way to set the log-level from a yaml file?

fujitatomoya, 2024-02-26:
there is a description in release note, https://docs.ros.org/en/rolling/Releases/Release-Humble-Hawksbill.html#passing-ros-arguments-to-node-actions.

it could be also added to documentation such as https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Creating-Launch-Files.html? (because release note is meant to show the change log and difference only.)

stewpend0us, 2024-02-27:
Thank you! Yes I think it should be mentioned in the 2nd link you shared. That's usually where I land when I'm setting up my launch file.

emersonknapp, 2025-11-12:
Being picked up by @lsy3 from ROSGraph WG discussion