ros2/ros2_documentation #4209: make linkcheck finds many `broken` links

fujitatomoya, 2024-03-05:
I think `ReleaseNote` can include the history, so some of the links are expected not to work any more, that is fine.
but tutorial and general pages should include the correct links available for good, at least we can check and fix the things return http error.

<details>
<summary>make linkcheck broken list</summary>

```bash
(Installation/Maintaining-a-Source-Checkout: line   10) broken    ../../rolling/Installation/Maintaining-a-Source-Checkout.html -
(Releases/Release-Galactic-Geochelone: line   45) broken    ../../../galactic/Installation.html -
(Releases/Release-Dashing-Diademata: line   35) broken    ../../dashing/Installation.html -
(Releases/Release-Eloquent-Elusor: line   35) broken    ../../eloquent/Installation.html -
(Releases/Release-Foxy-Fitzroy: line   32) broken    ../../foxy/Installation.html -
(Releases/Release-Humble-Hawksbill: line   44) broken    ../../humble/Installation.html -
(Releases/Release-Humble-Hawksbill: line  952) broken    ../../humble/Installation/Ubuntu-Install-Debians.html -
(Releases/Release-Iron-Irwini: line   45) broken    ../../iron/Installation.html -
(Concepts/Advanced/About-Middleware-Implementations: line   51) broken    http://www.omg.org/spec/DDS-XTypes - 500 Server Error: Internal Server Error for url: http://www.omg.org/spec/DDS-XTypes/
(Releases/Iron-Irwini-Complete-Changelog: line 1357) broken    https://build.ros2.org/view/Rbin_rhel_el864/job/Rbin_rhel_el864__mcap_vendor__rhel_8_x86_64__binary/ - 404 Client Error: Not Found for url: https://build.ros2.org/view/Rbin_rhel_el864/job/Rbin_rhel_el864__mcap_vendor__rhel_8_x86_64__binary/
(Tutorials/Advanced/Simulators/Webots/Installation-MacOS: line   49) broken    https://cdimage.ubuntu.com/focal/daily-live/pending/ - 404 Client Error: Not Found for url: https://cdimage.ubuntu.com/focal/daily-live/pending/
(Tutorials/Miscellaneous/Deploying-ROS-2-on-IBM-Cloud: line  468) broken    https://cloud.ibm.com/docs/containers?topic=containers-clusters#clusters_ui - Anchor 'clusters_ui' not found
(Tutorials/Miscellaneous/Deploying-ROS-2-on-IBM-Cloud: line  544) broken    https://cloud.ibm.com/docs/containers?topic=containers-images#namespace - Anchor 'namespace' not found
(Tutorials/Advanced/Simulators/Webots/Simulation-Supervisor: line   89) broken    https://cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_field_import_mf_node_from_string - Anchor 'wb_supervisor_field_import_mf_node_from_string' not found
(Tutorials/Advanced/Simulators/Webots/Simulation-Supervisor: line  109) broken    https://cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_node_remove - Anchor 'wb_supervisor_node_remove' not found
(How-To-Guides/Ament-CMake-Documentation: line  309) broken    https://github.com/ament/ament_lint/blob/rolling/ament_lint_auto/doc/index.rst#ament_lint_auto - Anchor 'ament_lint_auto' not found
(Releases/Humble-Hawksbill-Complete-Changelog: line 1510) broken    https://github.com/ros-visualization/qt_gui_core/tree/humble-devel/qt_gui_cpp/CHANGELOG.rst - 404 Client Error: Not Found for url: https://github.com/ros-visualization/qt_gui_core/tree/humble-devel/qt_gui_cpp/CHANGELOG.rst
(Releases/Release-Galactic-Geochelone: line  137) broken    https://github.com/ros2/demos/tree/a66f0e894841a5d751bce6ded4983acb780448cf/quality_of_service_demo#qos-overrides - Anchor 'qos-overrides' not found
(Releases/Release-Dashing-Diademata: line   96) broken    https://github.com/ros2/rcl_interfaces/blob/0aba5a142878c2077d7a03977087e7d74d40ee68/rcl_interfaces/msg/ParameterDescriptor.msg#L1 - Anchor 'L1' not found
(Releases/Iron-Irwini-Complete-Changelog: line   13) broken    https://github.com/ros2/rcl_interfaces/issues/215 - 404 Client Error: Not Found for url: https://github.com/ros2/rcl_interfaces/issues/215
(Releases/Release-Dashing-Diademata: line  739) broken    https://github.com/ros2/rclcpp/blob/ef41059a751702274667e2164182c062b47c453d/rclcpp_action/include/rclcpp_action/client.hpp#L399-L401 - Anchor 'L399-L401' not found
(Releases/Release-Dashing-Diademata: line  739) broken    https://github.com/ros2/rclcpp/blob/ef41059a751702274667e2164182c062b47c453d/rclcpp_action/include/rclcpp_action/client.hpp#L343 - Anchor 'L343' not found
(Releases/Release-Dashing-Diademata: line  739) broken    https://github.com/ros2/rclcpp/blob/ef41059a751702274667e2164182c062b47c453d/rclcpp_action/include/rclcpp_action/client.hpp#L276 - Anchor 'L276' not found
(Releases/Release-Dashing-Diademata: line  739) broken    https://github.com/ros2/rclcpp/blob/ef41059a751702274667e2164182c062b47c453d/rclcpp_action/include/rclcpp_action/client.hpp#L432-L434 - Anchor 'L432-L434' not found
(Tutorials/Advanced/Security/Examine-Traffic: line   30) broken    https://github.com/ros2/rmw_fastrtps#full-qos-configuration - Anchor 'full-qos-configuration' not found
(Releases/Release-Dashing-Diademata: line  766) broken    https://github.com/ros2/rviz/blob/dfceae319d49546f1e4ad39689853c18fef0001e/rviz_default_plugins/plugins_description.xml#L13 - Anchor 'L13' not found
(Releases/Iron-Irwini-Complete-Changelog: line 2746) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/12 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/12
(Releases/Iron-Irwini-Complete-Changelog: line 2744) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/14 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/14
(Releases/Iron-Irwini-Complete-Changelog: line 2743) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/15 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/15
(Releases/Iron-Irwini-Complete-Changelog: line 2745) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/13 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/13
(Releases/Iron-Irwini-Complete-Changelog: line 2741) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/17 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/17
(Releases/Iron-Irwini-Complete-Changelog: line 1359) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/31 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/31
(Releases/Iron-Irwini-Complete-Changelog: line 2742) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/16 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/16
(Releases/Iron-Irwini-Complete-Changelog: line 2760) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/3 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/3
(Releases/Iron-Irwini-Complete-Changelog: line 2755) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/5 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/5
(Releases/Iron-Irwini-Complete-Changelog: line 1360) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/32 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/32
(Releases/Iron-Irwini-Complete-Changelog: line 2753) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/6 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/6
(Releases/Iron-Irwini-Complete-Changelog: line 2747) broken    https://github.com/wep21/rosbag2_storage_mcap/issues/7 - 404 Client Error: Not Found for url: https://github.com/wep21/rosbag2_storage_mcap/issues/7
(Tutorials/Advanced/ROS2-Tracing-Trace-and-Analyze: line   67) broken    https://gitlab.com/ApexAI/performance_test/-/tree/master/performance_test#performance_test - Anchor 'performance_test' not found
(Releases/Galactic-Geochelone-Complete-Changelog: line  483) broken    https://importlib-metadata.readthedocs.io/en/latest/using.html - 404 Client Error: Not Found for url: https://importlib-metadata.readthedocs.io/en/latest/using.html
(Releases/Release-Eloquent-Elusor: line   47) broken    https://index.ros.org/doc/ros2/Tutorials/Getting-Started-With-Ros2doctor/ - 404 Client Error: Not Found for url: https://index.ros.org/doc/ros2/Tutorials/Getting-Started-With-Ros2doctor/
(Related-Projects: line   27) broken    https://index.ros.org/p/fmi_adapter/github-boschresearch-fmi_adapter/#foxy - Anchor 'foxy' not found
[... cut ...]
(How-To-Guides/Building-ROS-2-with-Tracing: line   36) broken    https://github.com/ros2/ros2_tracing#tracing - Anchor 'tracing' not found
(Releases/Release-Humble-Hawksbill: line  318) broken    https://github.com/ros2/sros2/blob/humble/SROS2_Linux.md#certificate-revocation-lists - Anchor 'certificate-revocation-lists' not found
(Tutorials/Advanced/Security/Access-Controls: line  143) broken    https://github.com/ros2/sros2/blob/rolling/sros2/test/policies/sample.policy.xml#L1 - Anchor 'L1' not found
```

</details>

clalancette, 2024-03-05:
> (Releases/Release-Crystal-Clemmys: line   37) broken    https://index.ros.org/r/rosbag2/github-ros2-rosbag2/#crystal - Anchor 'crystal' not found

At one point I was looking at turning this on for CI, but `linkcheck` has a problem with items like above.  That is a totally valid URL, but for whatever reason `linkcheck` cannot determine that the anchor is valid.  Until that is fixed somehow, we'll just need to keep manually running it and checking.

fujitatomoya, 2024-03-05:
@clalancette that is true, i did check some of the `broken` lines to make sure if linkcheck reports the actual broken URLs. some of them are totally no problem, as far as i checked we can exclude `anchor`, `local link`.

Argen1sw, 2025-03-03:
Hi @fujitatomoya, I want to contribute so I was wondering, is this issue still open? I saw the @m12watanabe1a proposed solution and the discussion in [#4487 ](https://github.com/ros2/ros2_documentation/pull/4487), but I am not quite sure if it was accepted or are you still looking for alternative solution to the problem.

fujitatomoya, 2025-03-03:
this issue is open, it would be appreciated if you could make a contribution. note that https://github.com/ros2/ros2_documentation/pull/4487 needs to address a couple of comments.

Argen1sw, 2025-03-06:
Hi @fujitatomoya
I have investigated the issue of Sphinx's linkcheck reporting broken links for anchors that are functional. It seems that this discrepancy often arises because some websites dynamically generate content using JavaScript, including anchors that aren't present in the initial HTML response (GitHub is one of these apparently). Since [linkcheck appears to analyze the static HTML](https://www.sphinx-doc.org/en/master/_modules/sphinx/builders/linkcheck.html#CheckExternalLinksBuilder), it might not detect these dynamically added anchors, leading to "Anchor not found" errors.​ Reference this thread for more details about the issue and how they were trying to fix it 4 years ago: https://github.com/sphinx-doc/sphinx/issues/9435

To address this, I propose the following solutions:

1. Disable Anchor Checking Globally (my recommendation): Set linkcheck_anchors = False (similar to the solution @m12watanabe1a proposed) in the Sphinx configuration file (conf.py). This setting instructs linkcheck to ignore all anchors while still verifying the main URLs. The advantage is that it confirms the base URL's validity without flagging anchor-related issues. However, the downside is that it won't detect broken anchors for any of the links.
`linkcheck_anchors = False`

2. Ignore Specific Anchors Using Regular Expressions: Utilize the linkcheck_anchors_ignore configuration option to specify regular expressions that match anchors Sphinx should skip when checking their validity. This approach is beneficial if certain anchors are known to be problematic or dynamically generated. However, it requires maintaining a list of regular expressions that accurately capture all such anchors across the project. For instance:
`linkcheck_anchors_ignore = [r'^!']`

3. Ignore Anchors for Specific URLs: Use the linkcheck_anchors_ignore_for_url configuration option to define regular expressions for URLs where anchor validation should be skipped. This method allows linkcheck to verify the main URL's accessibility while ignoring the anchors for specified URLs. I feel It will provide a more targeted approach but as well as 2. requires creating and maintaining regular expressions for URLs with dynamically generated anchors. For example:
`linkcheck_anchors_ignore_for_url = [r'https://github.com/.*']`

I have tested the first solution (linkcheck_anchors = False) as it is the most straightforward and ensures that the main URLs are checked without encountering anchor-related errors during the make linkcheck command. Below is a snippet of the test results, demonstrating that the main URLs are validated while anchors are ignored:​

**Github URL with deliberate typo and Anchor:**
`(Releases/Release-Crystal-Clemmys: line  128) broken    https://git.com/ros2/rcl/blob/657d9e84c73e4268176efd163e96fda73c1a76d9/rcl/include/rcl/context.h#L36-L136
`

**Github URL with no typo and Anchors:**
`(Releases/Release-Crystal-Clemmys: line  129) ok        https://github.com/ros2/rcl/blob/657d9e84c73e4268176efd163e96fda73c1a76d9/rcl/include/rcl/context.h#L138-L142
(Releases/Release-Crystal-Clemmys: line  131) ok        https://github.com/ros2/rcl/blob/657d9e84c73e4268176efd163e96fda73c1a76d9/rcl/include/rcl/context.h#L175-L205
(Releases/Release-Crystal-Clemmys: line  132) ok        https://github.com/ros2/rcl/blob/657d9e84c73e4268176efd163e96fda73c1a76d9/rcl/include/rcl/context.h#L207-L233
(Releases/Release-Crystal-Clemmys: line  133) ok        https://github.com/ros2/rcl/blob/657d9e84c73e4268176efd163e96fda73c1a76d9/rcl/include/rcl/context.h#L235-L255`

[full_output.txt](https://github.com/user-attachments/files/19098906/full_output.txt)

fujitatomoya, 2026-07-28:
@Firedeus thanks!

- 1st of all, i do not think https://github.com/ros2/ros2_documentation/pull/6176 is W.I.P anymore. no reply for a while. maybe we should close and replace it.
- yes, target branch is `rolling` 1st, and then we can work on backports to the downstream branches.
- about the anchor, it actually is working as valid URL, so it is useful for user experience. i guess we can keep `linkcheck_anchors = False` behavior.

vaatala, 2026-11-05:
It seems for Lyrical, many documentation links are broken, starting from:
https://docs.ros.org/en/rolling/Releases/Release-Lyrical-Luth.html#lyrical-luth-codename-lyrical-may-2026

**BROKEN**: "Install Lyrical Luth" -> https://docs.ros.org/en/lyrical/Installation.html

But, then if you go through here:
https://docs.ros.org/en/lyrical/Get-Started.html

This **works**: Installation -> https://docs.ros.org/en/lyrical/Get-Started/Installation.html