ros2/ros2_documentation #4213: colcon build not working

MonkeyBoy9999996, 2024-03-06:
# Summary
In a tutorial "Using colcon to build packages", the command `colcon build --symlink-install` does not work properly on my system. Cannot build it as tutorial states.

# Solution
I was able to find a solution. There are 2 following problems:
1. Tutorial should state that it would be a good practice to call it with parameter --parallel-workers N
  - default setting is to use maximum number of logical cores which in my case completely lagged the system
2. colcon fails to build with newer setuptools where it is deprecated to call setup.py directly
   - downgrading setuptools from version 68.2.2 to 58.1.0 worked
   - command `pip3 install setuptools==58.1.0`
   - ROS2 should make this working after instalation

# Enviroment
CPU: Intel Core i7-9750HQ 2.0Gh
RAM: 16GB
OS: Ubuntu 24.04.4 LTS 64bit

clalancette, 2024-03-06:
> 1. Tutorial should state that it would be a good practice to call it with parameter --parallel-workers N

This is a tradeoff.  First of all, there is no value of `N` we could put in there that would be correct.  It would be different for everyone.  Also, these are the beginner tutorials, so we want to make things as simple as possible.  So I don't particularly think we should do this.

> 2\. colcon fails to build with newer setuptools where it is deprecated to call setup.py directly

This should not have caused a failure.  It does cause warnings as you've seen, but they should not be fatal.  Can you please paste in the logs of the failing build?

MonkeyBoy9999996, 2024-03-06:
> This is a tradeoff. First of all, there is no value of `N` we could put in there that would be correct. It would be different for everyone. Also, these are the beginner tutorials, so we want to make things as simple as possible. So I don't particularly think we should do this.

I totally agree with that. Setting up a number of jobs is a more advanced topic. Yet still,  in my case my laptop lagged so much that I needed to forcely, with a power button, restart a computer. For a beginners tutorial, that can be quite discouraging. Further decision on that is on your judgment.

> This should not have caused a failure. It does cause warnings as you've seen, but they should not be fatal. Can you please paste in the logs of the failing build?

Starting >>> examples_rclpy_executors
--- stderr: examples_rclpy_executors                                                                                                                                                                       
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< examples_rclpy_executors [7.63s]
...
Summary: 22 packages finished [1min 53s]
  10 packages had stderr output: examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher launch_testing_examples

I agree with you it is a warning. Althoughts maybe a subsection describing a behaviour of that kind would be helpful. Again, it is up to your judgement.