ros2/ros2_documentation #4239: Nomenclature of tf_echo and tf_listener

lepalom, 2024-03-21:
I have a question about nomenclature that it is getting me crazy. Probably is because English is not my language.

The tool tf2_echo help says:

tf2_echo source_frame target_frame [echo_rate]

This will echo the transform from the coordinate frame of the source_frame
to the coordinate frame of the target_frame. 
Note: This is the transform to get data from target_frame into the source_frame.

For me is clear, if I have a frame, for instance, "grasp_01", and another frame "base", and I execute:
```
ros2 run tf2_ros tf2_echo base grasp_01
```

I would obtain the transformation that comes from base to grasp_01. Easily, I can see the distance between the origin of "base" to the origin of "grasp_01" expressed in the "base" frame.

But, in the [documentation](https://docs.ros.org/en/humble/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html#using-tf2-echo), for example the turtle, the sentence says:

>> Let’s look at the transform of the turtle2 frame with respect to turtle1 frame which is equivalent to:
>> ros2 run tf2_ros tf2_echo turtle2 turtle1

To me is the opposite, I got the transformation from turtle2 to turtle1, that is the same the frame turtle1 w.r.t. turtle2, not the opposite as claims the documentation.

Also, I got another issue, the tf_buffer class has the [method](https://docs.ros.org/en/ros2_packages/humble/api/tf2_ros/generated/classtf2__ros_1_1Buffer.html#classtf2__ros_1_1Buffer_1a3ab502cc1e8b608957a96ad350815aee)

lookupTransform (const std::string &target_frame, const std::string &source_frame, const tf2::TimePoint &time, const tf2::Duration timeout) const override

where you gets the transform between two frames by frame ID. With:
- target_frame = The frame to which data should be transformed
- source_frame = The frame where the data originated
...
 
If I create a listener in C++, and I call the function with:

tfs = tf_buffer_->lookupTransform("base", "grasp_01", tf2::TimePointZero);

I would obtain the same value as using tf2_echo. So, please, could you explain what should I understand as source_frame, target_frame  in the ROS context?

CursedRock17, 2024-04-09:
Duplicate of the following: 

https://github.com/ros/geometry/issues/108
https://github.com/ros/geometry/issues/111
https://github.com/ros/geometry/issues/226

clalancette, 2024-04-18:
I think what we need to do here is to make the target/source nomenclature completely consistent across the documentation, the tools, and the API.  @ahcorde I'm going to assign this one to you (along with #653) to eventually clean this up.