ros2/ros2_documentation #4277: Doc: numbering is off on page "Using variants"

VladimirFokow, 2024-04-15:
Here: https://docs.ros.org/en/humble/How-To-Guides/Using-Variants.html#creating-project-specific-variants

There are three points `1.` after each other. I don't think it makes sense.

In my opinion, it should be `1.`, `2.` and the third one - without a number at all.

christophebedard, 2024-04-15:
> In my opinion, it should be 1., 2. and the third one - without a number at all.

I agree. This is probably because the `code-block`s are not indented: https://github.com/ros2/ros2_documentation/blob/rolling/source/How-To-Guides/Using-Variants.rst?plain=1#L29-L57.

A PR to fix this would be welcome :smiley:

VladimirFokow, 2024-04-15:
should I also commit on humble branch?
what other branches?

clalancette, 2024-04-15:
> should I also commit on humble branch?

If you just make the change to the `rolling` branch, we'll use the bot to backport it as necessary.  Thanks!

PR #4278 (2024-04-15): fix numbering on page "Using variants"
closes #4277
commit 1ec3b5ce (2024-04-15): fix numbering on page "Using variants" (#4278)

* fix numbering on page "Using variants"

closes #4277
PR #4279 (2024-04-15): fix numbering on page "Using variants" (backport #4278)
closes #4277<hr>This is an automatic backport of pull request #4278 done by [Mergify](https://mergify.com).
commit 759586b3 (2024-04-15): fix numbering on page "Using variants" (#4278) (#4279)

* fix numbering on page "Using variants"

closes #4277

(cherry picked from commit 1ec3b5ce26defda9c26ebf7b5a17995260a28e89)
closed 2024-04-15 by clalancette