ros2/ros2_documentation #4297: confusion: does `colcon` or `rosdep` install packages from `package.xml`?

VladimirFokow, 2024-04-18:
[Here](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html#resolve-dependencies) (in section "4 Resolve dependencies") it says:

> Before building the workspace, you need to resolve the package dependencies. You may have all the dependencies already, but best practice is to check for dependencies every time you clone. You wouldn’t want a build to fail after a long wait only to realize that you have missing dependencies.
> 
> From the root of your workspace (ros2_ws), run the following command:
> ... `rosdep install` ...

<br />

But [here](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html#customize-package-xml) (in section "6 Customize package.xml") it says: 

> This is where your package.xml would list its dependencies on other packages, for **colcon** to search for

<br />

These 2 quotes seem to contradict each other.
Does `colcon` or `rosdep` install packages from `package.xml`?

clalancette, 2024-04-18:
I believe it is correct as-is.

`rosdep` is the tool that indeed walks the package.xml files and installs dependencies.

`colcon` is the tool that builds a workspace full of packages.  In order to do this, it has to know the topological ordering between them, so it *also* looks at the dependencies listed in package.xml to figure out that ordering.

If you have suggestions on how to improve the wording, we are of course happy to entertain them.

VladimirFokow, 2024-04-18:
Thank you for clarification!

I would replace "colcon" with "rosdep and colcon", just to be more clear.
But if there are reasons not to, I'm okay with closing this issue

MichaelOrlov, 2024-04-25:
@cottsay A friendly ping to follow up on this issue.