ros2/ros2_documentation #4302: Workshopping a ROS / Gazebo Tutorial Page to Land Under "Related Projects"

kscottz, 2024-04-19:
Quite a few users having a hard time figuring out how to make ROS 2 work with modern Gazebo. I propose we add a page under "related projects" that summarizes resources related to ROS / Gazebo integration. I intend to the solicit the community for resources they find useful. 

Here is an initial "strawman" list. 

* [Main Gz Docs](https://gazebosim.org/docs)
* [Gazebo Classic Migration](https://gazebosim.org/docs/harmonic/migrating_gazebo_classic_ros2_packages)
* [ROS / Gazebo Installation Guide](https://gazebosim.org/docs/harmonic/ros_installation)
* [ROS 2 / Gazebo Project Template](https://github.com/gazebosim/ros_gz_project_template)
* [ROS 2 / Gazebo Example Code](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim_demos)
* [Gazebo Community Forum](https://community.gazebosim.org/)
* [Gazebo Sim Forum](https://gazebosim.org/home)
* [Gazebo Source Code](https://github.com/gazebosim)
* [GazeboList.md](https://github.com/ros2/ros2_documentation/files/15045158/GazeboList.md)
* [Gazebosim Simulation Assets (aka Fuel)](https://app.gazebosim.org/fuel/models)
* [Gazebo Plugins](https://gazebosim.org/api/sim/7/createsystemplugins.html)

I've attached a list of other potential resources to go through that we think are worth adding. @RyanF says that he would be up for helping us reduce the attached list into a useful guide.

Ryanf55, 2024-04-19:
Thanks for this starter!

kscottz, 2024-04-19:
@Ryanf55 I think [this directory](https://github.com/ros2/ros2_documentation/tree/rolling/source/Related-Projects) would be a good place for this to land.  @clalancette has final say on location, and may have differing opinions.

Ryanf55, 2024-04-19:
LGTM for a central landing page. Only tutorials that are maintained should go there.

clalancette, 2024-04-22:
> I started by overhauling the `Gazebo.rst` page to have tabs for `Ignition` and `Garden or Newer`

This shouldn't be necessary.  Starting with Rolling/Jazzy, we are going to just be using a particular version of Gazebo with a particular version of ROS 2.  So we just need to update everything to use `gz`.  Also, we should also coordinate with https://github.com/ros2/ros2_documentation/pull/4311 and https://github.com/ros2/ros2_documentation/issues/3340 , where similar things are being discussed.

Ryanf55, 2024-04-22:
Subbed! I'm interested to ensure the tutorials work with `humble/harmonic` combination that ROS aerial is actively using. Per the last working group meeting, we're hoping to help improve official documentation on setup.

knmcguire, 2024-04-22:
Thanks for the headsup @Ryanf55! I was a bit too quick with creating [my draft PR](https://github.com/ros2/ros2_documentation/pull/4311) it seems for some improvements but I can close that and coordinate with you all. 

At least from the issue I was part of, the issue was that it was unclear what the recommended gazebo is for the distro from the preliminary of the documentation, as one doc says harmonic (gz) and the other says fortress (ign), so at least advising it to install it as dependency from ros-gz would ensure the right version for the ros bridge. 

but that being, if all will be updated to gz then maybe we should wait a bit indeed.

azeey, 2024-04-22:
I plan to bring this up in our Gazebo team meeting today, but my preference would be to have a minimal tutorial here showing how to install `ros_gz` with the assumption that they would use the officially paired version of Gazebo (e.g Humble/Fortress) and to point users to gazebosim.org for more in depth tutorials that discuss how to use Gazebo itself or how to set it up with different versions of Gazebo.

Ryanf55, 2024-04-22:
Sounds good! Perhaps this just needs to have a `gz -h` or `ign -h` and conclude the tutorial there.

knmcguire, 2024-04-23:
Alright just let us know what comes out of the meeting @azeey. From what you are explaining I can perhaps assume that [my PR](https://github.com/ros2/ros2_documentation/pull/4311) is more closer of what you had in mind, since that is what we discussed in the [issue](https://github.com/ros2/ros2_documentation/issues/3340) beforehand. It is only missing extra resources on the bottom (like those that @kscottz assembled above). But just let me know what you decide from the gazebo meeting.

azeey, 2024-04-23:
There was agreement at the meeting with what I said in https://github.com/ros2/ros2_documentation/issues/4302#issuecomment-2070261541. We can expand the ROS related tutorials in gazebosim.org/docs or create new ones as need be. I feel like there's still a lot of confusion as to which version of Gazebo should be paired with a version of ROS. Any clarifying content that y'all can add is most welcome.

Ryanf55, 2024-04-23:
You got it. I'll work with Kim.