ros2/ros2_documentation #4497: pip install fails with `error: externally-managed-environment`

fujitatomoya, 2024-05-29:
Using Ubuntu Nobel to build the documentation fails with the following error during installation of required packages.

```bash
root@tomoyafujita:~/ros2_ws/work/ros2_documentation# pip install -r requirements.txt -c constraints.txt
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

This is due to Ubuntu Nobel adopting [PEP 668 – Marking Python base environments as “externally managed”](https://peps.python.org/pep-0668/)

The followings are suggested fixes,

- https://github.com/ros2/ros2_documentation/blob/rolling/README.md, add virtualenv build example to avoid `--break-system-packages` option to override the system packages.
- https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html
  - update the procedure using virtualenv as well.
  - add how to use `devcontainer` to build the documentation. (https://github.com/ros2/ros2_documentation/blob/rolling/.devcontainer/devcontainer.json)

clalancette, 2024-05-31:
So the thing is, the README here already says that you need a virtualenv:

```
To build this you need to install

* make
* graphviz
* python virtualenv

In the virtualenv

``
pip install -r requirements.txt -c constraints.txt
``
```

I guess we can reword it to make it clearer, but I'm not sure what else to do here.

christophebedard, 2024-05-31:
Could we add the commands needed to create the venv? Before running `pip install`.

fujitatomoya, 2024-05-31:
@clalancette you can assign this to me.

closed 2024-10-08 by clalancette