ros2/ros2_documentation #4509: `Writing a simple service and client (Python)`: regular spin in the tuto

ymollard, 2024-06-04:
Hello,

I think that most beginners will use the official tuto to create their own service and client. This tuto calls the following method:
https://github.com/ros2/ros2_documentation/blob/dec345b136d0c3132cdf0561047e7012fe319bf5/source/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.rst?plain=1#L200

Although this works when the node is trivial and contains **only** a service call, most usecases will likely need other features requiring asynchronous calls (e.g. timer, publishers, subscribers...) that would create a deadlock if there is an existing `spin()` in the node.

I suggest to substitute `spin_until_future_complete` by `add_done_callback` + a regular `rclpy.spin()` in this tuto, so that beginners can copy/paste the code without risk of deadlock. It is still easy to understand for the tuto followers and also works for people copy/pasting code to their code.

What do you think about it? I can propose a modified version if you think it's good.

fujitatomoya, 2024-06-05:
Since this is beginner's tutorial and minimal example, i think it intentionally uses `spin_until_future_complete` to wait for the specific future. i think this also makes sense as a minimal tutorial and https://github.com/ros2/rclpy/blob/88ce194b723594ccfcb42aacc3bfc4c405bbde95/rclpy/rclpy/executors.py#L315 explains well.

I am not necessarily against changing this, but i really like to hear more feedback.

wjwwood, 2024-06-13:
We talked about this in the "waffle" issue triage meeting and the conclusion was that `spin_until_future_complete` is the right thing to use here, since that's the main purpose of this function is to help simplify situations like this example, but it would be a good idea to move the spin out of the `send_request()` method and into the main function, to better separate sending the request and waiting for it to complete. Also moving the spin activity to the main function will let them replace it in-place with spin if they decide to make it more complex.

PR #4538 (2024-06-13): Revamp the simple client tutorial to split the spin into main.
That way it is clearer that spin is a concept that is program wide, not just in the Node class.

This should fix #4509
PR #4540 (2024-06-13): Revamp the simple client tutorial to split the spin into main. (backport #4538)
That way it is clearer that spin is a concept that is program wide, not just in the Node class.

This should fix #4509<hr>This is an automatic backport of pull request #4538 done by [Mergify](https://mergify.com).
PR #4541 (2024-06-13): Revamp the simple client tutorial to split the spin into main. (backport #4538)
That way it is clearer that spin is a concept that is program wide, not just in the Node class.

This should fix #4509<hr>This is an automatic backport of pull request #4538 done by [Mergify](https://mergify.com).
PR #4542 (2024-06-13): Revamp the simple client tutorial to split the spin into main. (backport #4538)
That way it is clearer that spin is a concept that is program wide, not just in the Node class.

This should fix #4509<hr>This is an automatic backport of pull request #4538 done by [Mergify](https://mergify.com).
closed 2024-06-13 by clalancette