ros2/ros2_documentation #4523: Install ROS 2 on Jenson nano

prince-c11, 2024-06-11:
Please reply.
I am a beginner and I want to install ROS 2 Humble Hawksbill on Jenson nano. Since, Jenson nano works on ubuntu 18.04, so is there anyway to install Humble Hawksbill on Jenson nano.
If any other information needed please tell.

fujitatomoya, 2024-06-11:
Official supported platform is ubuntu 22.04, see more information for https://docs.ros.org/en/rolling/Releases/Release-Humble-Hawksbill.html#supported-platforms. (Tier 3 says ubuntu 20.04) But i have never seen humble on ubuntu 18.04, which is really old.

i know that this kind of constraints with BSP platforms, which is likely to happen. i guess that easiest and common way is to use containers ubuntu 22.04 and bind the hardware acceleration to the container. there are some documents provided by community, https://docs.ros.org/en/rolling/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html that you could use.

prince-c11, 2024-06-11:
Is there any way to install Ubuntu 22.04 on a Jetson Nano for using ROS 2 Humble Hawksbill?

fujitatomoya, 2024-06-11:
I believe that question should go to Nvidia Jetson Nano support, not application framework or libraries...

prince-c11, 2024-06-11:
Thank you.
Nvidia forum also said it is not supported.
[](https://forums.developer.nvidia.com/t/install-ros-on-jetson-nano/295728?u=prince.c11)

prince-c11, 2024-06-11:
[https://forums.developer.nvidia.com/t/install-ros-on-jetson-nano/295728?u=prince.c11](url)

fujitatomoya, 2024-06-11:
I guess that is not the question here to ask the platform support on specific BSP, can you close this issue and follow up with nvidia developer forum instead?

closed 2024-06-11 by prince-c11