ros2/ros2_documentation #4543: prebuilt html artifacts can help the review for each PR?

fujitatomoya, 2024-06-14:
### What is the pain?

During reviews, most likely at the very end of review, i want to see and check how updated html files are going to look like with incoming PR. and what i do now is to clone developer's repository and build it on my environment to see the built html files with the browser.
this happens several times for each PR, and that is kinda burden... i guess the same thing goes to all maintainers who are working on reviews.

### Proposal

Can we use github artifacts to upload the html files with github workflow in every commit? so that we can see html files with incoming changes only with downloading. No need to setup environment, clone, build the incoming PR repository.

### What's in it for us?

for developers, they can see the html artifacts built with proposed PR to verify.
for maintainers, it makes easier to check how the final html files look like with downloading the artifacts.

i think this could accelerate the review process for ROS 2 documentation.

clalancette, 2024-06-14:
> Can we use github artifacts to upload the html files with github workflow in every commit? so that we can see html files with incoming changes only with downloading. No need to setup environment, clone, build the incoming PR repository.

We used to have something like this where we would build and deploy to heroku.  But it was too much work to maintain so we gave up on it.

If we can figure out how to do this with a github artifact, that would be fine with me.  I think that would probably be easier to maintain than the heroku setup.

fujitatomoya, 2024-06-14:
thanks. not using any cloud services is one of the idea for maintenance, just relying on github action is fairly simple enough and i do not see any downside to add it.

closed 2024-06-14 by fujitatomoya