ros2/ros2_documentation #4548: Add artifact availability in contribution guide

fujitatomoya, 2024-06-14:
https://github.com/ros2/ros2_documentation/pull/4544 adds the html artifact in github action, it would be nicer to add this artifact is available to check the html built with incoming PR on [Contributing to ROS 2 Documentation](https://docs.ros.org/en/jazzy/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html).

html artifact is available in summary,

[image]

CC: @kscottz

kscottz, 2024-06-18:
@akashleena

christophebedard, 2024-07-10:
FYI this PR would make documenting this "feature" much simpler: https://github.com/ros2/ros2_documentation/pull/4564

fujitatomoya, 2024-07-21:
this has been addressed with https://github.com/ros2/ros2_documentation/pull/4559

closed 2024-07-21 by fujitatomoya