ros2/ros2_documentation #4558: Introducing-Turtlesim seems not to work (Beginner issue solved?)

FHefterSRH, 2024-06-26:
https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html

I had trouble to get the example to work in a ros2 docker image.

The problem was that on the web page the 
```sudo apt install ros-rolling-turtlesim``` 
addresses the wrong version. (For me)

This can be solved by simply replacing this line with:

```sudo apt install ros-${ROS_DISTRO}-turtlesim```

I know that for you it is obvious that the beginner is in the wrong version of the documentation.
 
But for a beginner, it is much easier to copy and paste the universal line above and continue playing without any problems.

What do you think?

christophebedard, 2024-06-26:
What ROS 2 distribution are you using? The expectation is that you're using the version of the docs corresponding to the distro you're using. For example, if you look at the Jazzy version of the tutorial, it uses `ros-jazzy-turtlesim`: https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html

christophebedard, 2024-06-26:
> I know that for you it is obvious that the beginner is in the wrong version of the documentation.
> 
> But for a beginner, it is much easier to copy and paste the universal line above and continue playing without any problems.

We cannot really have one version of a tutorial (or any page) that works for _all_ distros. That would be too complex to maintain.

christophebedard, 2024-06-26:
And for `${ROS_DISTRO}` to be set, you need to source the right file, so it would be tricky to include the `source ...` command everywhere in the docs.

christophebedard, 2024-06-27:
Since this is unfortunately not how the docs are set up, there is nothing to do, so I'll close this.

closed 2024-06-27 by christophebedard