ros2/ros2_documentation #4590: Two issues with Docker tutorial in Jazzy

DaveCollinsJr, 2024-08-02:
My Environment: Ubuntu Jammy 22.04.4 LTS
Docker version 24.0.7, build 24.0.7-0ubuntu2~22.04.1

Attempting to run ros:jazzy via the Docker Guide [Setup ROS2 with VSCode and Docker Container](https://docs.ros.org/en/jazzy/How-To-Guides/Setup-ROS-2-with-VSCode-and-Docker-Container.html#id9)

I see two problems with the jazzy Setup-ROS-2-with-VSCode-and-Docker-Container instructions:

**1) The supplied UID 1000 already exists**
When you build the container you receive the error:
`groupadd: GID '1000' already exists`
Simply changing USER_UID to 1001 in the Dockerfile solves this problem.

**2) The DISPLAY env var does not seem to be being passed to the container**
In the devcontainer.json the containerEnv.DISPLAY value does not seem to carry over to the container.
When it is set to the suggested "DISPLAY": "unix:0",
in my container:
```
dave@DROIDBASE1:/home/ws$ echo $DISPLAY
:0
```
Then I try setting it to "unix:1" and rebuild the container, no change to $DISPLAY:
```
dave@DROIDBASE1:/home/ws$ echo $DISPLAY
:0

```
The other containerEnv vars seem fine. Is there possibly something wrong with the runArgs command?

Consequently rviz2 cannot be made to run in the container, but I suspect that is related to the DISPLAY issue.

If someone confirms the USER_UID=1001 is an appropriate fix I will make a PR to the docs to fix that issue.

fujitatomoya, 2024-08-05:
@DaveCollinsJr if you can review https://github.com/ros2/ros2_documentation/pull/4591, that would be really appreciated.

fujitatomoya, 2024-08-05:
> 2) The DISPLAY env var does not seem to be being passed to the container
> In the devcontainer.json the containerEnv.DISPLAY value does not seem to carry over to the container.

this is expected behavior, `runArgs` overwrite that during initializing container from the host system setting. if you comment out that `"-e", "DISPLAY=${env:DISPLAY}"`, `containerEnv.DISPLAY` will be set.

DaveCollinsJr, 2024-08-06:
Thanks much @fujitatomoya :+1: . Your #4591 changes solved the duplicate UID problem.  Also thanks for the note about commenting out the "-e" line. That does, indeed, properly set $DISPLAY in the container.

I'm still having RVIZ2 problems but will troubleshoot that separately and open a different ticket if I determine it is a documentation issue.

DaveCollinsJr, 2024-08-06:
For anyone reading this having rviz2 display issues (e.g.: "qt.qpa.xcb: could not connect to display unix:0") following the Docker tutorial:

I was able to resolve this by executing the ```xhost local:root``` command on my local Ubuntu machine before creating the docker container. The response to that command should be something along the lines of: ```non-network local connections being added to access control list```

fujitatomoya, 2024-08-06:
> For anyone reading this having rviz2 display issues (e.g.: "qt.qpa.xcb: could not connect to display unix:0") following the Docker tutorial:

Nice catch, probably it would be nice to add this here as well.

> I was able to resolve this by executing the xhost local:root command on my local Ubuntu machine before creating the docker container.

i think we are creating `ros2dev` local user here in container, so `xhost +local` should work and disallowing any remote hosts.

closed 2024-08-20 by clalancette