ros2/ros2_documentation #4595: ros2 service echo is not recognised!

PeymanAmirii, 2024-08-07:
Hi. I use ROS2 Iron on Ubuntu 22. All the commands for topics and services mentioned in the tutorial work correctly except for `ros2 service echo`. The terminal shows the following error:
`invalid choice: 'echo' (choose from 'call', 'find', 'list', 'type')`

It seems that the choice `echo` is not recognised by ros2 iron for the command `ros2 service`.
This command has been discussed in the tutorial here:
https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Services/Understanding-ROS2-Services.html#ros2-service-echo

PeymanAmirii, 2024-08-07:
Thank you. I am very interested in ROS2 and test everything mentioned in the tutorial one by one. Is there any opportunity for me to join the team of ROS2 developers? I have a broad background in robotics and programming.

clalancette, 2024-08-07:
> Thank you. I am very interested in ROS2 and test everything mentioned in the tutorial one by one. Is there any opportunity for me to join the team of ROS2 developers? I have a broad background in robotics and programming.

The best thing to do to get involved is to continue running and testing things, reporting issues, and opening pull requests.  If you have enough contributions, then over time you may apply to be mentored into the ROS core developers.

PeymanAmirii, 2024-08-08:
Thanks Chris!

fujitatomoya, 2024-08-08:
https://github.com/ros2/ros2_documentation/pull/4594 is merged, closing this issue.

@PeymanAmirii thanks for posting the issue!

PR #4594 (2024-08-08): Update the documentation about service echo for Iron.
In Iron, we did not have the dedicated "ros2 service echo" command.  You can still see the results of the service using a "ros2 topic echo", so put that in the documentation instead.

Fixes #4595

@fujitatomoya FYI
closed 2024-08-08 by fujitatomoya