ros2/ros2_documentation #4610: Topics & Services

PeymanAmirii, 2024-08-10:
Hi.
I have read and used the commands mentioned in the tutorial of ROS2 Iron for topics and services. The following suggestions came to my mind which make the commands for topics and services more rememberable and consistent.

For topics:
1. It is beneficial to create the command of `ros2 topic find <topic_type>` like what we have for services as `ros2 service find <setvice_type>` which finds all the services of a specific type.

2. The command `ros2 topic info <topic_name>` just shows the number of publishers and subscribers. It is beneficial that this command lists the names of publisher and subscriber nodes of a topic, as well. Like the command `ros2 node info <node_name>` which lists the names of all the topics, services, and actions connected to a node.

For services:
3. It is beneficial to create the command of `ros2 service info <service_name>` like what we have for topics as `ros2 topic info <topic_name>` which shows the number of publishers and subscribers of a topic.

Best wishes,
Peyman

fujitatomoya, 2024-08-20:
> It is beneficial to create the command of ros2 topic find <topic_type> like what we have for services as ros2 service find <setvice_type> which finds all the services of a specific type.

`ros2 topic fine` already does this as following.

```bash
# Terminal-1
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
...
# Terminal-2
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub -r 1 /greetings std_msgs/msg/String "{data: 'Hello from terminal'}"
...
# Terminal-3
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic find std_msgs/msg/String
/chatter
/greetings
```

> The command ros2 topic info <topic_name> just shows the number of publishers and subscribers. It is beneficial to mention on the tutorial that by using the option --verbose

probably we do not want to mention all options, so we could add something like `see more options with --help` for the tutorial.

> It is beneficial to create the command of ros2 service info <service_name> like what we have for topics as ros2 topic info <topic_name> which shows the number of publishers and subscribers of a topic.

we already have this in rolling and jazzy with https://github.com/ros2/ros2cli/pull/771, not available on iron or humble.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 service info /add_two_ints
Type: example_interfaces/srv/AddTwoInts
Clients count: 0
Services count: 1
```

verbose option: https://github.com/ros2/ros2cli/pull/916 is W.I.P.

> It is useful to add an option to the command ros2 node info <node_name> to show the namespace of the node.

i do not really understand the requirement here, can you elaborate a bit?
user should already know the namespace when calling this command since `<node_name>` requires namespace.
otherwise, `Unable to find node '<node_name>'` will return.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 node info talker
Unable to find node 'talker'
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 node info /talker
/talker
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
...
```

PeymanAmirii, 2024-08-21:
Hi Fujitatomoya,
Thanks for your reply. 

> `ros2 topic fine` already does this as following.
I checked `ros2 topic find <topic_type> ` and yes, it exists. But nothing about it has been mentioned on the topic tutorial of ROS2 Iron at:
https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html
Please add it.

> probably we do not want to mention all options, so we could add something like `see more options with --help` for the tutorial.
Yes, please add this explanation for all distributions.

> we already have this in rolling and jazzy with [ros2/ros2cli#771](https://github.com/ros2/ros2cli/pull/771), not available on iron or humble.
Good to hear that it has been added for the newer version.

> i do not really understand the requirement here, can you elaborate a bit?
> user should already know the namespace when calling this command since `<node_name>` requires namespace.
node name and node namespace are two different things. For example, when we want to run a create a node, we can use the following command to run a node with a specific node name and namespace.
`ros2 run <package_name> <executable_name> --ros-args --remap __node:=<node_name> -r __ns <node_name_space>`
However, the command ros2 node info <node_name>, does not show the node namespace.

Best wishes,
Peyman

fujitatomoya, 2024-08-21:
> we can use the following command to run a node with a specific node name and namespace.

can you share the `following command`?

> the command ros2 node info <node_name>

do you mean it should print `<fully qualified node name>` instead?

PeymanAmirii, 2024-08-21:
Yes, you are right. When a user uses the command `ros2 node info <node namepace/node_name>` already knows the namespace. 

As a summary of the suggestions, the command `ros2 topic find <topic_type>` and the option `--help` for the command `ros2 topic info <topic_name>` and the input  `<node namepace/node_name>` for the command `ros2 node info <node namepace/node_name>` can be added to the tutorials.

fujitatomoya, 2024-08-21:
> the command ros2 topic find <topic_type>

https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html does not introduce the all `ros2 topic` sub-commands. I think that is okay to add more, i would love to review the PR. (`bw`, `delay` and `find` can be added.)

> option --help for the command ros2 topic info <topic_name>

https://docs.ros.org/en/rolling/Concepts/Basic/About-Command-Line-Tools.html introduces `ros2 --help` and that clearly tells:

```
...
Call `ros2 <command> -h` for more detailed usage.
```

I do not think we need to add this explanation of `--help` option in the tutorial.

> the input <node namepace/node_name> for the command ros2 node info <node namepace/node_name> can be added to the tutorials.

i would argue that this should be explained in `ros2node` cli but documentation. so that user can realize that we need to specify the fully qualified node name including namespace here.

see https://github.com/ros2/ros2cli/pull/923

mnhaqq, 2024-10-02:
> 
> https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html does not introduce the all `ros2 topic` sub-commands. I think that is okay to add more, i would love to review the PR. (`bw`, `delay` and `find` can be added.)

I would like to contribute to this issue and introduce more `ros2 topic` sub-commands. I will start working on it and submit a PR soon. Please let me know if there's anything specific I should keep in mind

commit a839c3b0 (2024-10-15): Update Understanding-ROS2-Topics.rst (#4803)

* Update Understanding-ROS2-Topics.rst

Introduce bw and find commands to understanding topics tutorial

* Update Understanding-ROS2-Topics.rst

Rephrase description of bw command for clarity

Fixes #4610

* Update Understanding-ROS2-Topics.rst

Remove redundant lines

* Update Understanding-ROS2-Topics.rst

* Update Understanding-ROS2-Topics.rst

Changes for readability and remove unnecessary lines
commit 7a6f4247 (2024-10-15): Update Understanding-ROS2-Topics.rst (#4803) (#4820)

* Update Understanding-ROS2-Topics.rst

Introduce bw and find commands to understanding topics tutorial

* Update Understanding-ROS2-Topics.rst

Rephrase description of bw command for clarity

Fixes #4610

* Update Understanding-ROS2-Topics.rst

Remove redundant lines

* Update Understanding-ROS2-Topics.rst

* Update Understanding-ROS2-Topics.rst

Changes for readability and remove unnecessary lines

(cherry picked from commit a839c3b0faee22a8ac9f0737293ed1dd7eb2ccf7)
commit a13897bc (2024-10-15): Update Understanding-ROS2-Topics.rst (#4803) (#4819)

* Update Understanding-ROS2-Topics.rst

Introduce bw and find commands to understanding topics tutorial

* Update Understanding-ROS2-Topics.rst

Rephrase description of bw command for clarity

Fixes #4610

* Update Understanding-ROS2-Topics.rst

Remove redundant lines

* Update Understanding-ROS2-Topics.rst

* Update Understanding-ROS2-Topics.rst

Changes for readability and remove unnecessary lines

(cherry picked from commit a839c3b0faee22a8ac9f0737293ed1dd7eb2ccf7)
commit 8f27b460 (2024-10-15): Update Understanding-ROS2-Topics.rst (#4803) (#4818)

* Update Understanding-ROS2-Topics.rst

Introduce bw and find commands to understanding topics tutorial

* Update Understanding-ROS2-Topics.rst

Rephrase description of bw command for clarity

Fixes #4610

* Update Understanding-ROS2-Topics.rst

Remove redundant lines

* Update Understanding-ROS2-Topics.rst

* Update Understanding-ROS2-Topics.rst

Changes for readability and remove unnecessary lines

(cherry picked from commit a839c3b0faee22a8ac9f0737293ed1dd7eb2ccf7)
closed 2024-10-15 by clalancette