ros2/ros2_documentation #4706: tutorial ERROR!

Meg-ZC, 2024-09-01:
in [this section](https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.rst), the code name _add_two_ints_server.cpp_ got wrong when i was building the package.
It turns out that the 8th line rclcpp::node::SharedPtr should be rclcpp::Node::SharedPtr

fujitatomoya, 2024-09-01:
@Meg-ZC thanks for posting issue.

and yes good eye 👁️ , https://github.com/ros2/ros2_documentation/pull/4707 is the fix.

Meg-ZC, 2024-09-02:
> @Meg-ZC thanks for posting issue.
> 
> and yes good eye 👁️ , #4707 is the fix.

fail to build this package

ahcorde, 2024-09-02:
The PR is merged, feel free to reopen the issue if the problem persists

closed 2024-09-02 by ahcorde