ros2/ros2_documentation #4893: Missing Dependency rti-connext-dds-6.0.1 in rosdep Installation while building jazzy from source

meetgandhi-dev, 2024-12-08:
Hey,

I hope you're doing well. I noticed that the dependency `rti-connext-dds-6.0.1` is not being installed when running rosdep to install dependencies for jazzy, as referenced [here](https://github.com/ros2/ros2_documentation/blob/jazzy/source/Installation/Alternatives/Ubuntu-Development-Setup.rst#install-dependencies-using-rosdep). Since it is listed as a required dependency in the corresponding [package.xml](https://github.com/ros2-gbp/rmw_connextdds-release/blob/release/jazzy/rti_connext_dds_cmake_module/0.22.0-2/package.xml#L13), I believe it should be included in the installation process. This seems to be causing issues when generating standalone Debian packages.

If this was done intentionally, I would be more than happy to submit a pull request to fix it. :)
Thank you for your time and consideration!

christophebedard, 2024-12-08:
I believe this is intentional. RTI Connext DDS is not the default `rmw` implementation, and [installing it requires accepting its license](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations.html#rti-connext-version-6-0-1-amd64-only) and then [setting an environment variable before building `rmw_connextdds`](https://github.com/ros2/rmw_connextdds#quick-start) (otherwise it gets skipped). Therefore, if you want it, you should indeed explicitly install it.

meetgandhi-dev, 2024-12-08:
> I believe this is intentional. RTI Connext DDS is not the default `rmw` implementation, and [installing it requires accepting its [...]

Oh! That makes sense. This helps clarify things, thank you so much!

closed 2024-12-08 by meetgandhi-dev