ros2/ros2_documentation #4901: add explanation about component container types and options

fujitatomoya, 2024-12-13:
found on https://github.com/ros2/rclcpp/pull/2541#discussion_r1883133428, it is really hard for user to know the component container options.

we could add the following to the documentation, https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Composition.html
- general explanation about component container, which is the node that loads/unloads the components and execute them.
- component container types and options, such as `component_container`, `component_container_mt` with parameter `thread_num`, and `component_container_isolated` with `--use_multi_threaded_executor` option.

fujitatomoya, 2024-12-13:
@clalancette @christophebedard @ahcorde what do you think?

christophebedard, 2024-12-13:
it should absolutely be documented, but maybe in a tutorial rather than a _concepts_ article? I think this is the only composition tutorial: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Composition.html

clalancette, 2024-12-13:
> it should absolutely be documented, but maybe in a tutorial rather than a _concepts_ article? I think this is the only composition tutorial: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Composition.html

Yeah, agreed.  The general explanation about the component container can go into the Concepts, but then I think we should expand the tutorial instead.

christophebedard, 2024-12-13:
The first bullet point in this section of the composition concepts article does mention "generic container process" and links to `component_container.cpp`: https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Composition.html#using-components. We could just briefly mention here that there are other containers and link to the tutorial.

In the tutorial, `component_container` is used. We could add a new subsection under _Advanced Topics_ to list the other containers and their options: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Composition.html#advanced-topics.

closed 2024-12-26 by clalancette