ros2/ros2_documentation #4913: Ubuntu 22.04 is listed as supported platform for Jazzy

apirek, 2024-12-20:
[The docs](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#system-requirements) list Ubuntu 22.04 as a supported OS for ROS 2 Jazzy.
[REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029) does not.

Is Ubuntu 22.04 really a Tier 3 supported OS? So far I found that some of navigation2's packages don't build on it because of the yaml-cpp version provided by the OS.

clalancette, 2024-12-20:
> [The docs](https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html#system-requirements) list Ubuntu 22.04 as a supported OS for ROS 2 Jazzy.
> [REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029) does not.

Good point.  The docs are correct, and Ubuntu 22.04 is indeed a Tier-3 supported platform for Jazzy.  We just forgot to update REP-2000 for that.  I'll submit a PR to update it.

clalancette, 2024-12-20:
OK, I've opened up https://github.com/ros-infrastructure/rep/pull/413, which should fix it.

Even though that hasn't been merged yet, I'm going to close this since GitHub isn't great about closing issues across organizations.  Thanks for the report.

closed 2024-12-20 by clalancette

apirek, 2024-12-20:
Thank you for clarifying.