ros2/ros2_documentation #4924: Add a tutorial on Lifecycle nodes

Yadunund, 2025-01-06:
Extend the `Intermediate` tutorials with a chapter on implementing Lifecycle nodes and designing APIs using `NodeInterfaces` such that the APIs work with both regular Nodes and LifecycleNodes. The tutorial could also include launch events to transition lifecycle states from launch files.

kscottz, 2025-01-07:
Just for clarity here, you are asking for an example node that implements both the [standard node interface](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/node_interfaces/node_interfaces.hpp) and the [lifecycle node interface](https://github.com/ros2/rclcpp/blob/rolling/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp). 

Roughly this means that the standard node interface performs the primary state's / transitions when it is instantiated as a standard node? If this is the case, it might be helpful to have a paster template that people can re-use as a starting point. 

Do you have a good example of `NodeInterfaces` being used with a lifecycle node out in the wild? Even tests might be helpful here.

Yadunund, 2025-01-09:
I was alluding to following the pattern described by @methylDragon in his [lightning talk](https://docs.google.com/presentation/d/1bdXOOZPhR9yAnyGNxoLhuO_bU4RW5IjnzvCtrVlpe_g/edit#slide=id.g286ccf0c527_0_4).

This would encourage more users to use `LifecycleNodes` while keeping other APIs compatible with both `Node` and `LifecycleNode`.

kscottz, 2025-01-09:
Perfect. Thanks. I can work with this! I may even be able to find a new contributor to do it.

NickTziaros, 2025-01-25:
> I may even be able to find a new contributor to do it.

Is there a way I can be of assistance? Im not an expert by any means but i would like to get my hands dirty with this one.

kscottz, 2025-01-27:
@NickTziaros by all means! Give it a shot.

Make sure to take a look at the [docs readme](https://github.com/ros2/ros2_documentation/blob/rolling/README.md) and the [contributing guide](https://docs.ros.org/en/jazzy/The-ROS2-Project/Contributing.html). Make sure to put your editor into whitespace mode as the linter is really particular.

kscottz, 2025-02-11:
Thanks for the help @NickTziaros.

Yadunund, 2025-02-12:
So we're still missing a tutorial on implementing lifecycle nodes, their various `on_*` methods, transitioning states via cli commands & launch actions. @NickTziaros's addition can be extended to include these topics. Should we keep this ticket open or open a separate one?

fujitatomoya, 2025-02-12:
@Yadunund good eye, let's reopen this.

Yadunund, 2025-02-12:
I've updated the description accordingly 👍🏼

kscottz, 2025-02-12:
Good catch @Yadunund. Sorry about that. 

On the other hand we really need to work towards smaller, faster pull requests and this fits that pattern.

NickTziaros, 2025-02-15:
> So we're still missing a tutorial on implementing lifecycle nodes, their various on_* methods, transitioning states via cli commands & launch actions

I might as well make this one too :D

NickTziaros, 2025-02-16:
I made a PR #5038, adding
```diff
+Output:
+ 
```
above

```
[INFO] [Simple_Node]: Node name: Simple_Node 
[INFO] [Simple_LifeCycle_Node]: Node name: Simple_LifeCycle_Node
```
for consistency. Somehow I missed it the first time.

NickTziaros, 2025-03-12:
I noticed that there is already a quite thorough tutorial on  Lifecycle nodes [code](https://github.com/ros2/demos/blob/rolling/lifecycle/README.rst) in the Demo section under [Managing nodes with managed lifecycles](https://docs.ros.org/en/rolling/Tutorials/Demos/Managed-Nodes.html). Wouldn't it be redundant if we added another?
Additionaly, the link [rosindex](https://index.ros.org/p/lifecycle/github-ros2-demos/) is broken.

kscottz, 2025-03-12:
@NickTziaros I think we're fine with a little redundancy but we should cross reference the two. 

With respect to ROS Index we've been doing some work to improve index search and remove redundant package listings. [I think this is the package that we should have there. ](https://index.ros.org/p/lifecycle/)

inventrX, 2026-08-31:
Yo @Yadunund  @kscottz , I'd like to take this on. 

I just started technical writing and I've just been setting up ROS2 Jazzy on Ubuntu 24.04. 
I can write the tutorial covering:
1. Implementing on_configure, on_activate, on_deactivate with code examples
2. Transitioning states via CLI: `ros2 lifecycle set /node configure` 
3. Doing transitions in launch files
4. Cross-linking to the existing Demo tutorial to avoid redundancy.

I'm not an expert or anything but I feel this will be a really good learning experience for me.
I’ll follow the docs style guide.

Yadunund, 2026-09-09:
@inventrX sounds good! Feel free to tag me when it's ready for a review.