ros2/ros2_documentation #4927: Parameter namespace usage

wouter-heerwegh, 2025-01-08:
I was recently converting my ROS1 yaml files to ROS2 and we had the following structure:

```yaml
car:
  motor:
    max_torque: 5 #Nm
```

Since this variable is used in multiple nodes, I converted this to the following:

```yaml
/**:
  ros__parameters:
    car:
      motor:
        max_torque: 5 #Nm
```

Accessing this variable however was not that straight forward. I declared the parameter in my code as follows:
```c++
handle->declare_parameter<float>("car/motor/max_torque", max_torque);
```

But this didn't seem to work. I was trying to look through the documentation and  the design documentation but couldn't find out how I could read from a sub namespace or parameter namespace until I found [this reddit thread](https://www.reddit.com/r/ROS/comments/11vnssg/how_to_define_and_access_subnamespaces_in_ros_2/)

Changing my code to the following seems to work
```c++
handle->declare_parameter<float>("car.motor.max_torque", max_torque);
```

I couldn't really find an explanation on how to do this in the documentation. Is this already part of the documentation? Or should a section be added to further clarify?

fujitatomoya, 2025-01-08:
I think it would be nice to add the parameter name section somewhere in the documentation. (as far as i see, there is no such explanation...)

https://design.ros2.org/articles/ros_parameters.html

> Address all parameters without ambiguity in the names
> For example, one of the challenges of the current system is that there is a naming ambiguity between nodes and parameters /foo/bar/baz could be a node /foo/bar/baz or a private parameter baz on node /foo/bar

i believe this is major reason.

besides, just a reference.

https://github.com/ros2/rclcpp/blob/9cabd69412eb2e66f93af73c614b4db25f7c7501/rclcpp/include/rclcpp/node.hpp#L506-L515

kscottz, 2025-01-09:
The crux of your issue, if I understand correctly, is that we don't have a tutorial on simply using parameters (i.e. how to create a parameter and set its value in a YAML param file).

Skimming the docs I see:
* [using `ros2 param`](https://docs.ros.org/en/rolling/How-To-Guides/Using-ros2-param.html)
* [Using a param (C++/Python)](https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html)
* [Checking for param updates](https://docs.ros.org/en/rolling/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-CPP.html)

None of those show how to create a file and access its value. I believe this should be part of the "Using a Param" guide. 

This is a great first issue. Let me see if I can find a new contributor to help us out.

jkaniuka, 2025-01-12:
Hi @kscottz,

I've submitted PR :wink:

closed 2025-01-14 by kscottz

jkaniuka, 2025-01-14:
Thanks everyone, glad I could contribute :robot: