ros2/ros2_documentation #4959: ROS2 migration guide replaces spinOnce with spin_some

wouter-heerwegh, 2025-01-23:
I recently stumbled on an issue where I was using `spin_some` in a node that was receiving messages at a higher rate than it was calling the `spin_some` function. After a bunch of digging, I found the following issue on rclcpp ros2/rclcpp#2655, which mentions that `spin_some` only processes 1 message of each callback, rather then all work on all callbacks at that time.

It seems that `spin_all` functionality more closely resembles the `spinOnce` behaviour from ROS1.

My code was converted according to the [migration guide](https://docs.ros.org/en/jazzy/How-To-Guides/Migrating-from-ROS1/Migrating-CPP-Package-Example.html#changing-c-library-calls), so I think it would be best to change the example to use `spin_all` instead

kscottz, 2025-01-23:
@wouter-heerwegh thanks for reporting this. 

@fujitatomoya I want a quick second opinion, does this sound like a reasonable request? If the answer is yes I can put together a PR.

Somewhat related, the [spin_all API documentation](https://docs.ros.org/en/jazzy/p/rclcpp/generated/function_namespacerclcpp_1af0d7db8ca1fc6ea5884e79a96b8abaf9.html) could use some love and this might warrant a second ticket.

fujitatomoya, 2025-01-23:
@kscottz i guess, some semantic for them are different from ROS 1 to ROS 2, but including that i will check to see if the doc or implementation needs to be updated. i will take a look.

fujitatomoya, 2025-01-24:
@wouter-heerwegh 

> spin_some only processes 1 message of each callback, rather then all work on all callbacks at that time.

precisely `spin_some` can process a data for each entity. (if 2 subscriptions are data-ready, it calls both user callbacks.)

IMO, the doc section is not well described... i would expect that all ready data in the single entity will be executed. that is the claim from https://github.com/ros2/rclcpp/issues/2655.

https://github.com/ros2/rclcpp/blob/2d1b770e858fdb20e9c25913341fb388100dd19e/rclcpp/include/rclcpp/executor.hpp#L283-L303

> It seems that spin_all functionality more closely resembles the spinOnce behaviour from ROS1.

I do not think so.

https://docs.ros.org/en/diamondback/api/roscpp/html/namespaceros.html#af4afa6f0ad9f903f04a023982a95ff1c says `Process a single round of callbacks` and processes only one message from each callback queue in the node handle's associated callback queue during a single call. If you have two subscriptions and call ros::spinOnce() once, it will process only one callback per subscription (if messages are available for both).

that said, `ros::spinOnce()` matches to `Executor::spin_some`, not `Executor::spin_all`.

after all, i think migration doc is correct. what do you think?

wouter-heerwegh, 2025-01-24:
@fujitatomoya 
Thanks for the reply.

While it's definitely defined as such, I think the behaviour is slightly different. When I check the implementation of `spinOnce` [in the roscpp documentation](https://docs.ros.org/en/noetic/api/roscpp/html/init_8cpp_source.html#l00585), it seems to call `callAvailable`. And that function is documented as if it would process all callbacks. So it looks a bit like `spin_all` with a timeout of 0.

I'll make a small subscriber and publisher node to test the behaviour in ROS1, just to confirm.

kscottz, 2025-01-24:
I think the real issue here, as show by this issue and #2655,  is that the behavior of `spin`, `spin_some`, and `spin_all` isn't self-evident and the API docs don't really clarify the difference.  

Regardless of where we land on this particular problem, we probably want to improve the API documentation and write a guide explaining when to use each variant of `spin`.

fujitatomoya, 2025-01-25:
> I'll make a small subscriber and publisher node to test the behaviour in ROS1, just to confirm.

that definitely helps here 🚀 thanks!

wouter-heerwegh, 2025-01-25:
@fujitatomoya, I did the following tests:
- ROS1 `spinOnce` implementation
- ROS2 `spin_some` implementation
- ROS2 `spin_all` implementation with 0 max_duration
- ROS2 `spin_all` implementation with 5s max_duration

# ROS1 `spinOnce`
### Subscriber
```c++
#include "ros/ros.h"
#include "std_msgs/String.h"

ros::Time last_msg_time;

void chatterCallback(const std_msgs::String::ConstPtr &msg)
{
	ROS_INFO_STREAM("I heard: [" << msg->data.c_str() << "], time offset: " << (ros::Time::now() - last_msg_time).toSec());
	last_msg_time = ros::Time::now();
}

int main(int argc, char **argv)
{
	ros::init(argc, argv, "listener");
	ros::NodeHandle n;

	ros::Subscriber sub = n.subscribe("chatter", 1000, chatterCallback);
	last_msg_time = ros::Time::now();
	ros::Rate loop_rate(0.2);
	while(ros::ok())
	{
		ros::spinOnce();
		loop_rate.sleep();
		ROS_INFO_STREAM("Looping...");
	}

	return 0;
}
```

### Publisher
```c++
#include "ros/ros.h"
#include "std_msgs/String.h"

#include <sstream>

int main(int argc, char **argv)
{
  ros::init(argc, argv, "talker");
  ros::NodeHandle n;
  ros::Publisher chatter_pub = n.advertise<std_msgs::String>("chatter", 1000);

  ros::Rate loop_rate(1);

  int count = 0;
  while (ros::ok())
  {
    std_msgs::String msg;

    std::stringstream ss;
    ss << "hello world " << count;
    msg.data = ss.str();

    ROS_INFO("%s", msg.data.c_str());

    chatter_pub.publish(msg);

    ros::spinOnce();

    loop_rate.sleep();
[... 247 lines cut ...]
[INFO] [1737835723.998086001] [minimal_publisher]: Publishing: 'Hello, world! 3'
[INFO] [1737835724.997963379] [minimal_publisher]: Publishing: 'Hello, world! 4'
[INFO] [1737835725.998044965] [minimal_publisher]: Publishing: 'Hello, world! 5'
[INFO] [1737835726.998007839] [minimal_publisher]: Publishing: 'Hello, world! 6'
[INFO] [1737835727.998002223] [minimal_publisher]: Publishing: 'Hello, world! 7'
[INFO] [1737835728.998010047] [minimal_publisher]: Publishing: 'Hello, world! 8'
[INFO] [1737835729.997992773] [minimal_publisher]: Publishing: 'Hello, world! 9'
[INFO] [1737835730.997988944] [minimal_publisher]: Publishing: 'Hello, world! 10'
[INFO] [1737835731.997993312] [minimal_publisher]: Publishing: 'Hello, world! 11'
[INFO] [1737835732.997991355] [minimal_publisher]: Publishing: 'Hello, world! 12'
[INFO] [1737835733.998011520] [minimal_publisher]: Publishing: 'Hello, world! 13'
[INFO] [1737835734.998001647] [minimal_publisher]: Publishing: 'Hello, world! 14'
[INFO] [1737835735.997984794] [minimal_publisher]: Publishing: 'Hello, world! 15'
[INFO] [1737835736.998004819] [minimal_publisher]: Publishing: 'Hello, world! 16'
[INFO] [1737835737.998000897] [minimal_publisher]: Publishing: 'Hello, world! 17'
[INFO] [1737835738.998006518] [minimal_publisher]: Publishing: 'Hello, world! 18'
[INFO] [1737835739.998002386] [minimal_publisher]: Publishing: 'Hello, world! 19'
[INFO] [1737835740.997965316] [minimal_publisher]: Publishing: 'Hello, world! 20'
[INFO] [1737835741.997967480] [minimal_publisher]: Publishing: 'Hello, world! 21'
[INFO] [1737835742.997893886] [minimal_publisher]: Publishing: 'Hello, world! 22'
```

# Conclusion

From these tests it does look as if `spin_all` is more closely resembling the `spinOnce` behaviour.

fujitatomoya, 2025-01-26:
> From these tests it does look as if spin_all is more closely resembling the spinOnce behaviour.

@wouter-heerwegh thank you very much for confirming this! and based on this result, i agree with you conclusion.

there is a couple of things that i need to do,

- i will check if this ROS 2 `rclcpp::spin_some` behavior is by design with other maintainers. if not, we need to fix `rclcpp::spin_some` implementation. otherwise, we need to update the doc section to clarify the behavior.
- even though `rclcpp::spin_some` needs to be fixed, current migration is not correct for user at this moment. ROS 1 `spinOnce` should be replaced with ROS 2 `rclcpp::spin_some` as we confirmed here.

fujitatomoya, 2025-01-26:
@wouter-heerwegh @kscottz 

could you take a look at https://github.com/ros2/ros2_documentation/pull/4966?

wouter-heerwegh, 2025-01-26:
Looks good to me, I guess another issue should be opened to keep track of a more elaborate part of the documentation explaining the differences between the spin methods, right?

If some help is needed, I would gladly help out.

fujitatomoya, 2025-01-27:
thanks!

> I guess another issue should be opened to keep track of a more elaborate part of the documentation explaining the differences between the spin methods, right?

i can do that once i confirm the designed behavior.

closed 2025-01-27 by ahcorde