ros2/ros2_documentation #4968: Clarification needed: Should we avoid using services?

karaketir16, 2025-01-27:
The documentation contains two statements that seem related:  

https://github.com/ros2/ros2_documentation/blob/f9b32d55caaddb9a5431da4ccbca34ed4e77e67e/source/How-To-Guides/Sync-Vs-Async.rst?plain=1#L29
https://github.com/ros2/ros2_documentation/blob/f9b32d55caaddb9a5431da4ccbca34ed4e77e67e/source/How-To-Guides/Topics-Services-Actions.rst?plain=1#L24

1. In [Sync-Vs-Async.rst](https://github.com/ros2/ros2_documentation/blob/f9b32d55caaddb9a5431da4ccbca34ed4e77e67e/source/How-To-Guides/Sync-Vs-Async.rst?plain=1#L29):  
   "Avoid sync calls"  

2. In [Topics-Services-Actions.rst](https://github.com/ros2/ros2_documentation/blob/f9b32d55caaddb9a5431da4ccbca34ed4e77e67e/source/How-To-Guides/Topics-Services-Actions.rst?plain=1#L24):  
   "Services are simple blocking calls"  

**Does this mean we should avoid using services?**  

I think the documentation's explanation of services seems to suggest they are intended for synchronous calls.
```
* Should be used for remote procedure calls that terminate quickly, e.g. for querying the state of a node or doing a quick calculation such as IK. They should never be used for longer running processes, in particular processes that might be required to preempt if exceptional situations occur and they should never change or depend on state to avoid unwanted side effects for other nodes.
* Simple blocking call. Mostly used for comparably fast tasks as requesting specific data. Semantically for processing requests.
```

karaketir16, 2025-01-27:
I understand that we can use asynchronous service calls via ``call_async()``. However, if we use this approach, the service call is no longer a ``simple blocking call`` as described in the documentation.

clalancette, 2025-01-27:
> **Does this mean we should avoid using services?**

No, not at all.  We would provide services if they should be avoided.

> I understand that we can use asynchronous service calls via `call_async()`. However, if we use this approach, the service call is no longer a `simple blocking call` as described in the documentation.

I agree that the phrase "Simple blocking call" might be confusing.  For what it is worth, it is definitely possible to use it in a simple blocking way, though it takes two lines of code.  In any case, my suggestion is just to remove that bullet point completely.  Would you mind opening a pull request to do that?

karaketir16, 2025-01-27:
> For what it is worth, it is definitely possible to use it in a simple blocking way, though it takes two lines of code.

If you are using a single threaded executor and do it in a callback, you can cause a deadlock.(If you did not specify timeout it is deadlock. If you use future.wait_for(timeout), it is not a deadlock, but never gets a response, always gives timeout).

As described [Here](https://github.com/ros2/ros2_documentation/blob/482b754eba5d8bc071b62f45029cf46d2e6fa404/source/How-To-Guides/Using-callback-groups.rst?plain=1#L164C1-L180C1)
```
With the above in mind, here are a couple guidelines to help avoid deadlocks:

* If you make a synchronous call in any type of a callback, this callback and
  the client making the call need to belong to

  * different callback groups (of any type), or
  * a Reentrant Callback Group.

* If the above configuration is not possible due to other requirements - such
  as thread-safety and/or blocking of other callbacks while waiting for the
  result (or if you want to make absolutely sure that there is never a
  possibility of a deadlock), use asynchronous calls.

Failing the first point will always cause a deadlock.
An example of such a case would be making a synchronous service call
in a timer callback (see the next section for an example).
```

You need to use multi threaded executor and callback groups as in `Solution`.
https://github.com/ros2/ros2_documentation/blob/482b754eba5d8bc071b62f45029cf46d2e6fa404/source/How-To-Guides/Using-callback-groups.rst?plain=1#L435

I think it is not a `simple blocking way`.

In any case, I've opened a pull request as you recommended. Thank you for your very quick response.

clalancette, 2025-01-27:
> If you are using a single threaded executor and do it in a callback, you can cause a deadlock.

Correct.  That is the main anti-pattern we want to address here.  But there are many other ways to do this, including using a multi-threaded executor and callback groups, and also putting a spin into another thread.

PR #4970 (2025-01-27): Remove confusing services definition
Removed a services bullet point as suggested in https://github.com/ros2/ros2_documentation/issues/4968#issuecomment-2615509697

closes #4968
PR #4971 (2025-01-27): Remove confusing services definition (backport #4970)
Removed a services bullet point as suggested in https://github.com/ros2/ros2_documentation/issues/4968#issuecomment-2615509697

closes #4968<hr>This is an automatic backport of pull request #4970 done by [Mergify](https://mergify.com).
closed 2025-01-27 by fujitatomoya