ros2/ros2_documentation #4984: Strategies for YAML Message Generation Using ROS CLI

kscottz, 2025-01-31:
[image]

This one comes in from the Discord, but it is also something I've encountered personally. For complex ROS messages it can be difficult to use the ROS CLI. I think it would be helpful to have an intermediate tutorial on how to dump a YAML message from the CLI, modify the message, and then republish the message using `ros2 topic pub` etc. 

I am pretty sure there is a way to do this but it is non-obvious to new users. It would be helpful to have something to refer to for debugging.

fujitatomoya, 2025-01-31:
@kscottz 

just wanted to share current status.

currently we can `ros2 topic echo` to save the messages in yaml file format, but we cannot `ros2 topic pub` with using yaml file just yet.
the original issue https://github.com/ros2/ros2cli/issues/919, and enhancement is on review https://github.com/ros2/ros2cli/pull/925.

adding documentation, i totally agree once the feature is ready.

kscottz, 2025-02-01:
Thanks for adding context @fujitatomoya. I seemed to remember that we had this discussion earlier in the year. I'll see if I can't help move along that PR next week. Once that gets merged this would be a fantastic task for a first time contributor.

fujitatomoya, 2025-02-07:
https://github.com/ros2/ros2cli/pull/925 has been merged in rolling. (this feature is available on rolling only.)

CC: @kscottz

kscottz, 2025-02-07:
Perfect. Thanks @fujitatomoya. I just added this issue to push list for next week. I'll see if we can't find a new contributor to help work this out.

NickTziaros, 2025-02-11:
Can I take this one as well?

fujitatomoya, 2025-02-11:
@NickTziaros absolutely, thanks for the effort! if you have any questions, let me know. and i am happy to review!

PR #5034 (2025-02-17): added Publishing-Messages-Using-YAML-Files.rst and edited Intermediat…
Closes #4984 

I created a first draft of the tutorial.

I used a geometry_msgs/msg/Twist message for the demo, assuming that those new to ROS 2 are already familiar with it from the Understanding Topics tutorial. However, since this functionality is especially useful for editing and publishing more complex structured messages, I could switch to using a geometry_msgs/msg/PoseStamped message instead.

@fujitatomoya  Let me know what you think and if this is what you had in mind.
closed 2025-02-17 by fujitatomoya