ros2/ros2_documentation #4986: Add a warning to action doc that it cannot be created under ament_python

EmmanuelMess, 2025-02-02:
The https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html document should have a warning that only a C++ package can create an action. Messages and services have the same restriction AFAIK, but I haven't check if they say so. I think this is not readily apparent, and might lead to confusion as most of the docs are separated into "how to do in python" and "how to do in cpp", but actions have no such separation and only can be done in C++.

I dont know if there is an explicit recommendation to separate interfaces into their own packages, but it should probably be added to the action creation docs, if it makes sense.

clalancette, 2025-02-02:
> The https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Creating-an-Action.html document should have a warning that only a C++ package can create an action. Messages and services have the same restriction AFAIK, but I haven't check if they say so.

To be clear, it is not that it has to be a C++ package, but rather that it has to be a CMake package; the language doesn't matter.  And we do indeed say that it has to be a CMake package in https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html#create-a-new-package .  So I'd be in favor of adding similar language to https://docs.ros.org/en/rolling/Tutorials/Intermediate/Creating-an-Action.html

clalancette, 2025-02-02:
Oh, and in case it wasn't clear; we'd very much appreciate if you opened a PR adding it!

EmmanuelMess, 2025-02-03:
> To be clear, it is not that it has to be a C++ package, but rather that it has to be a CMake package; the language doesn't matter. 

Wait, does that mean that you can create "crossed" cmake python packages? Is that documented somewhere I can read?

> Oh, and in case it wasn't clear; we'd very much appreciate if you opened a PR adding it!

I am doing a graduate thesis right now, so that consumes most of my time, but if I can find the time I will fix these things.

clalancette, 2025-02-03:
> Wait, does that mean that you can create "crossed" cmake python packages? Is that documented somewhere I can read?

Yes, it is possible.  The documentation for it is in https://docs.ros.org/en/rolling/How-To-Guides/Ament-CMake-Python-Documentation.html .  That said, we don't really recommend using it much; it is better to use `ament_cmake` for C++ and for message/service/action packages, and `ament_python` for python packages.

Yadunund, 2025-02-15:
@EmmanuelMess leaving this ticket open for you or anyone else to submit a PR to update the documentation.

vimal0athithan, 2025-03-04:
hello team,
can I take over resolving this issue?

Yadunund, 2025-03-04:
@vimal0athithan absolutely!

kscottz, 2025-03-10:
@vimal0athithan how is it going? Are you making any progress? Do you need some help?

vimal0athithan, 2025-03-30:
Hello @EmmanuelMess,@Yadunund

I've created PR [#5201](https://github.com/ros2/ros2_documentation/pull/5201) to address this issue by adding explicit warnings about action creation limitations. As a new contributor (@vimal0athithan), I'd appreciate your feedback on the implementation.

The changes include:

Added a clear warning that actions require ament_cmake (not ament_python)

Included a note about interface separation best practices

Positioned the warnings before implementation steps for maximum visibility

This should help prevent confusion, especially for Python-focused developers. Please let me know if you'd like any adjustments to the wording or placement.

(Note: My apologies for the delayed response - I was temporarily hospitalized but wanted to complete this contribution as my first ROS documentation PR.)

vimal0athithan, 2025-03-30:
> [@vimal0athithan](https://github.com/vimal0athithan) how is it going? Are you making any progress? Do you need some help?

thank you so much for offering.