ros2/ros2_documentation #4988: Doxygen is missing for newer ros versions

EmmanuelMess, 2025-02-03:
I am trying to find the jazzy version of https://docs.ros2.org/foxy/api/tf2/ But I only find http://docs.ros.org/en/jazzy/p/tf2/generated/index.html#. I also find that I have the same problem with other packages (for example rclcpp).

Where is the doxygen? Some of the information is available at sphinx but it is much more difficult to read, and graphs are entirely missing.

clalancette, 2025-02-03:
> Where is the doxygen?

We switched from doxygen into sphinx, to have a more unified view of all of the documentation (not just the doxygen).  There is some work going on over at https://github.com/ros-infrastructure/rosdoc2 to improve it.

Given that this is a question, not a request for change, I'm going to close this out.  But please feel free to keep commenting.

closed 2025-02-03 by clalancette

EmmanuelMess, 2025-02-03:
> We switched from doxygen into sphinx, to have a more unified view of all of the documentation (not just the doxygen). There is some work going on over at [ros-infrastructure/rosdoc2](https://github.com/ros-infrastructure/rosdoc2?rgh-link-date=2025-02-03T13%3A44%3A03.000Z) to improve it.

I don't understand, rosdoc2 allows for the creation of doxygen, but it won't be created anymore for any packages?

clalancette, 2025-02-03:
> I don't understand, rosdoc2 allows for the creation of doxygen, but it won't be created anymore for any packages?

rosdoc2 generated doxygen, then converts it into rst files so it can be integrated with the rest of the sphinx site for each individual package.  So the same data that doxygen generates is available, it is just presented in a different way.

https://docs.ros2.org is a legacy site, just there for historical purposes.  No new documentation will go there anymore, and it will eventually be decommissioned.

EmmanuelMess, 2025-02-03:
>  So the same data that doxygen generates is available, it is just presented in a different way.

Yes, but it is much more difficult to navigate (missing graphs, different search, different layout, etc), especially if you are already accustomed to navigating doxygen pages. Are the pages not generated anymore or just not published?

EmmanuelMess, 2025-02-03:
Is there any way we can have both until breathe/exhale has feature parity with doxygen pages?

clalancette, 2025-02-03:
> Is there any way we can have both until breathe/exhale has feature parity with doxygen pages?

No, it is too much effort for us to maintain.  But we welcome help on making breathe/exhale have feature parity!