ros2/ros2_documentation #5004: Requesting ROS 2 words for codespell ROS 2 dictionary support.

fujitatomoya, 2025-02-08:
follow-up of https://github.com/ros2/ros2_documentation/pull/4997

currently codespell checker uses ony [builtin dictionaries](https://github.com/codespell-project/codespell/tree/main/codespell_lib/data), but we could have ROS 2 dictionary to let the codespell get them for more documentation quality.

I tried some experimental things like below, and i found that `ignore-words` cannot work `case-sensitive`. that does not let the codespell detect the words from the dictionary.
see more details for https://github.com/codespell-project/codespell/issues/3638.

```patch
diff --git a/Makefile b/Makefile
index e8a706bb..d783a687 100644
--- a/Makefile
+++ b/Makefile
@@ -29,7 +29,7 @@ test:
        doc8 --ignore D001 --ignore-path build
 
 spellcheck:
-       git ls-files '*.md' '*.rst' | xargs codespell --ignore-words=codespell_whitelist.txt --skip="source/Releases/*"
+       git ls-files '*.md' '*.rst' | xargs codespell -I codespell_whitelist.txt -D codespell_dictionary.txt,- --skip="source/Releases/*"
 
 linkcheck:
        $(BUILD) -b linkcheck $(OPTS) $(SOURCE) $(LINKCHECKDIR)
diff --git a/codespell_dictionary.txt b/codespell_dictionary.txt
new file mode 100644
index 00000000..1c1fb1c1
--- /dev/null
+++ b/codespell_dictionary.txt
@@ -0,0 +1 @@
+ROS2->ROS 2
diff --git a/codespell_whitelist.txt b/codespell_whitelist.txt
index b7860a05..e2a3b1ba 100644
--- a/codespell_whitelist.txt
+++ b/codespell_whitelist.txt
@@ -1 +1,2 @@
 empy
+ros2
```

what the above modification wants to do is to catch `ROS2` and suggest `ROS 2`(with space) but `ros2`. this is because we have many words `ros2` for command line examples such as `ros2 topic echo xxx`.
but this is not working because of https://github.com/codespell-project/codespell/issues/3638.

**after all, what we can do at this moment is that `case-insensitive` spell check and suggestion for ROS 2.**
**if you have any idea or suggestions, please post it here. i am all ears** 👂👂👂

fujitatomoya, 2025-02-12:
`ros2bag->rosbag2`

peci1, 2025-02-12:
tpoic->topic,srevice->service

Or these already contained in the general spellchecker?

amnet->ament,amemt->ament

fujitatomoya, 2025-02-12:
thanks guys.

> Or are these already contained in the general spellchecker?

@peci1 no i do not think so, but i can check for that.

> Depending on the scope of the dictionary you wish to create, so I’ll give an example for now.

atm general one.

> Words that I think are commonly used throughout ROS.

@Tacha-S those can be added to ignore word instead, i guess. unless you found misspelling examples before like `ros2bag->rosbag2`?

christophebedard, 2025-02-12:
> `ros2bag->rosbag2`

What if the docs mention the [`ros2bag` package](https://github.com/ros2/rosbag2/tree/rolling/ros2bag)?

Tacha-S, 2025-02-13:
I thought it would be something like creating a dictionary for ROS 2, similar to this:  
https://github.com/streetsidesoftware/cspell-dicts/tree/main/dictionaries  

The ROS 2 example makes sense, but considering misspellings, there’s no end to the possibilities—such as swapping one character within a known word, adding an extra character, or omitting a character. I doubt it’s possible to cover everything.  

Therefore, I assumed this dictionary would focus only on proper nouns related to ROS 2. However, in the example, **ros2bag -> rosbag2** is listed.  

In my opinion, *ros2bag* might have been intended to be corrected to *rosbag2*, but it could also refer to a command like *ros2 bag play*, so it’s unclear what the purpose is.  

If we were to give similar examples, it might look like:  
- Moveit2 -> MoveIt 2
- Ros -> ROS
- Rmw -> RMW
- Nav 2 -> Nav2

fujitatomoya, 2025-02-13:
> I thought it would be something like creating a dictionary for ROS 2, similar to this:

no it is more like https://github.com/codespell-project/codespell/tree/main/codespell_lib/data

> there’s no end to the possibilities

yeah, understood. probably it would be useful what happened before, i guess.

> ros2bag might have been intended to be corrected to rosbag2

as i posted above, there is an actual package for ros2cli extension, called ros2bag. so this is my bad example... sorry for the confusion.

some of your examples make sense, i will give it a shot.

as described in the issue header, actually there is an issue for the codespell, https://github.com/codespell-project/codespell/issues/3638. this blocks us to detect `ROS2->ROS 2` because `ros2` is the correct word.

anyway, thank you very much for your idea! appreciate that.

fujitatomoya, 2025-03-03:
> i found that ignore-words cannot work case-sensitive. that does not let the codespell detect the words from the dictionary.
see more details for https://github.com/codespell-project/codespell/issues/3638.

i am working to fix codespell for this requirement https://github.com/codespell-project/codespell/pull/3658.
once we can get it merged and released from codespell project, we can detect `ROS2` with `ROS 2` suggestion.

CC: @kscottz @christophebedard