ros2/ros2_documentation #5075: automatic spell check documentation

fujitatomoya, 2025-03-10:
### Overview

automatic spell check has been added with https://github.com/ros2/ros2_documentation/pull/4997 and https://github.com/ros2/ros2_documentation/pull/5058 for ROS 2 documentation.
This issue is to add the documentation how to use automatic spell check, how to add/remove the words to/from the dictionaries. So that developers can run the spell check locally before opening the pull request, and we can also expect the users and developers can add more possible miss-spelling words to our ROS 2 specific dictionaries easily.

### Consideration

- this doc could land around https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html
- introduction how codespell works with ROS 2 documentation with ignore/miss-spell dictionaries.
- explain how to run spell check locally and fix them.
- explain how to add/remove the suggestions to the dictionaries.

fujitatomoya, 2025-03-10:
CC: @kscottz

SuperGops7, 2025-03-10:
@fujitatomoya can I take this up?

kscottz, 2025-03-10:
@SuperGops7 if you think you are up for it I'll assign it to you. 

Just to be clear, what we would like you to do is add a short section to this [part of the docs](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html) explaining how to use the codespell feature. 

Tomoya listed a pretty good summary above. A couple things to keep in mind:

1. Get the docs building locally before you begin.
2. Remember one sentence per line and turn on whitespace mode  (our linter is really picky about trailing whitespace).
3. If you run into problems please comment here or take it to Discord. 
4. If you get busy and can't finish please let us know.

SuperGops7, 2025-03-11:
@kscottz got it, no worries
I already have a local build running from one of my previous PRs.

PR #5105 (2025-03-18): Update documentation for automatic spellcheck
Update the docs for automatic spellcheck.

Fixes #5075
PR #5124 (2025-03-18): Update documentation for automatic spellcheck (backport #5105)
Update the docs for automatic spellcheck.

Fixes #5075 <hr>This is an automatic backport of pull request #5105 done by [Mergify](https://mergify.com).
PR #5125 (2025-03-18): Update documentation for automatic spellcheck (backport #5105)
Update the docs for automatic spellcheck.

Fixes #5075 <hr>This is an automatic backport of pull request #5105 done by [Mergify](https://mergify.com).
closed 2025-03-18 by fujitatomoya