ros2/ros2_documentation #5120: symlink-install on Windows 11 requires admin or Developer Mode. This should be in the colcon tutorial.

CraigBuilds, 2025-03-17:
The colcon tutorial mentions symilnk install, but this will fail for Windows 11 users: https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#build-the-workspace

This can be fixed by enabling developer mode. This should be added to the documentation on symlink-install.

https://github.com/colcon/colcon-core/issues/689

mjcarroll, 2025-03-28:
@CraigBuilds this sounds like like it would be a fantastic documentation improvement.  Would you be interested in adding it?  If so, I would be happy to review.

CraigBuilds, 2025-03-28:
I'm swamped with work until April 8th, but I can attempt to do this after that :)

I have never contributed to an open source project before, nor contributed to documentation, so it will be a learning process.

If I've not done it by the end of April, it's likely I've become swamped with work again and this slipped my mind.

mjcarroll, 2025-03-28:
Would definitely appreciate it if you have time! 

The place you are looking is: https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.rst