ros2/ros2_documentation #5139: ROS2 Humble Windows Binary Installation - newest version of empy has breaking changes

FlufferOverflow, 2025-03-20:
In the [Windows 10 binary installation instructions for Humble](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#install-dependencies), empy is listed as one of the dependencies to be installed with pip. As of the time of writing, this will install version 4.2 of empy, which breaks rosidl\_adapter and ros2pkg:
```
G:\dev\Ceres\ProtogenFaceTracking\ROS>colcon build --merge-install
Starting >>> turtlesim
--- stderr: turtlesim
CMake Error at C:/dev/ros2_humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(C:/Python38/python.exe -m rosidl_adapter --package-name
  turtlesim --arguments-file
  G:/dev/Ceres/ProtogenFaceTracking/ROS/build/turtlesim/rosidl_adapter__arguments__turtlesim.json
  --output-dir
  G:/dev/Ceres/ProtogenFaceTracking/ROS/build/turtlesim/rosidl_adapter/turtlesim
  --output-file
  G:/dev/Ceres/ProtogenFaceTracking/ROS/build/turtlesim/rosidl_adapter/turtlesim.idls)
  returned error code 1:

  AttributeError processing template 'action.idl.em'

  Traceback (most recent call last):

    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\resource\__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,

  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "C:\Python38\lib\runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "C:\Python38\lib\runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\__main__.py", line 19, in <module>
      sys.exit(main())
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\__init__.py", line 29, in convert_to_idl
      return convert_action_to_idl(
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\action\__init__.py", line 39, in convert_action_to_idl
      expand_template('action.idl.em', data, output_file, encoding='iso-8859-1')
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\resource\__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "C:\dev\ros2_humble\Lib\site-packages\rosidl_adapter\resource\__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  C:/dev/ros2_humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:24 (rosidl_generate_interfaces)

---
Failed   <<< turtlesim [13.0s, exited with code 1]

Summary: 0 packages finished [14.0s]
  1 package failed: turtlesim
  1 package had stderr output: turtlesim
```
```
G:\dev\Ceres\ProtogenFaceTracking\ROS\src>ros2 pkg create --build-type ament_cmake core_tensorrt
going to create a new package
package name: core_tensorrt
destination directory: G:\dev\Ceres\ProtogenFaceTracking\ROS\src
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['FlufferOverflow']
licenses: ['TODO: License declaration']
build type: ament_cmake
dependencies: []
creating folder .\core_tensorrt
creating .\core_tensorrt\package.xml
Traceback (most recent call last):
  File "C:\dev\ros2_humble\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_humble\Lib\site-packages\ros2cli\cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2_humble\Lib\site-packages\ros2pkg\command\pkg.py", line 37, in main
    return extension.main(args=args)
  File "C:\dev\ros2_humble\Lib\site-packages\ros2pkg\verb\create.py", line 184, in main
    create_package_environment(package, args.destination_directory)
  File "C:\dev\ros2_humble\Lib\site-packages\ros2pkg\api\create.py", line 99, in create_package_environment
    _create_template_file(
  File "C:\dev\ros2_humble\Lib\site-packages\ros2pkg\api\create.py", line 81, in _create_template_file
    _expand_template(template_path, template_config, output_file_path)
  File "C:\dev\ros2_humble\Lib\site-packages\ros2pkg\api\create.py", line 31, in _expand_template
    em.BUFFERED_OPT: True,
AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
```
The [ROS Index page](https://index.ros.org/p/ros2pkg/#humble-deps) for ros2pkg indicates that the package [python3-empy](https://packages.ubuntu.com/jammy/python3-empy) is a dependency, which is version 3.3.4 of empy. Installing that version of empy with pip instead of version 4.2 fixes the problem with rosidl\_adapter and ros2pkg.

christophebedard, 2025-03-20:
Thanks for reporting this! Your analysis makes sense.

> The [ROS Index page](https://index.ros.org/p/ros2pkg/#humble-deps) for ros2pkg indicates that the package [python3-empy](https://packages.ubuntu.com/jammy/python3-empy) is a dependency, which is [...]

According to REP 2000, Humble on Windows targets EmPY version 3.3.2: https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027. Would you be willing to open a PR against the `humble` branch to constrain `empy`?

FlufferOverflow, 2025-03-25:
The fix was merged, so I'm going to go ahead and close this issue.

closed 2025-03-25 by FlufferOverflow