ros2/ros2_documentation #5188: 📝 Use URL links to the demo/example source code instead of pasting it in doc section.

fujitatomoya, 2025-03-28:
### Issue Type

- [ ] 🐛 Bug / Problem
- [ ] ✏️ Typo / Grammar
- [ ] 📖 Outdated Content
- [x] 🚀 Enhancement

### Distribution

Humble, Jazzy, Rolling

### Description

In the documentation, there are demo source codes that are basically copy&paste from either [examples](https://github.com/ros2/examples) or [demos](https://github.com/ros2/demos).

If the implementation in the documentation is required to explain how it works in details for the user step by step, it needs to be in the documentation. But if that only shows the code and demonstrates it to see the output, probably the source code does not need to be copied in the documentation, instead we can use URL links to the original example or demo source code.

this enhances the maintainability for the documentation, to avoid the possible conflict between the demo codes in the documentation and the code from original repository.

### Suggested Fix

Use the URL links to the demo and example code if possible.

For example, https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Demos/Content-Filtering-Subscription.rst

### Additional Context

This came up from https://github.com/ros2/ros2_documentation/pull/5184

fujitatomoya, 2025-03-28:
CC: @christophebedard

christophebedard, 2025-03-28:
> But if that only shows the code and demonstrates it to see the output, probably the source code does not need to be copied in the documentation, instead we can use URL links to the original example or demo source code.

Exactly. And if the demo source code does change, I wouldn't expect it to really make the corresponding tutorial/demo article invalid.